betaflight/betaflight-configurator #1877: Add Pilot Name to the Configurator UI

bizmar, 2020-02-09:
BF has Craft Name and Pilot Name as an option in the CLI and separate OSD elements, it would be nice if I could set Pilot Name in the Personalization part of the Configuration tab. May need to be added to MSP first.

etracer65, 2020-02-11:
To be clear, it's the "display name" not pilot name.

But this will be more difficult than it would appear. The display name is currently not included in the MSP. Also the preexisting MSP messages where it would make sense to add the display name were designed in a legacy way and cannot be added to for this. So it would take new messages to get/set the display name in the firmware.

mikeller, 2020-02-11:
On the plus side, we support MSPv2 now, so we are not limited to 256 messages any more.

PR #2798 (2022-10-30): Add 'Pilot name' to the Configurator UI; rename 'Display name' to 'Pilot name'; rename 'name' to 'craft_name'
Fixes #1877

Related firmware changes @ https://github.com/betaflight/betaflight/pull/11391

---

**scope**

This PR renames:
- the `Display name` OSD / configuration element (and the internal `FC.CONFIG.displayName`) to `Pilot name` (hence `FC.CONFIG.pilotName`).
- the internal  `FC.CONFIG.name` to `FC.CONFIG.craftName` (with no functional impact)

It also adds the option to set the `pilot name` through the configurator UI. The value can be show in the OSD (independently of the curren [... cut]
closed 2022-10-30 by haslinghuis