betaflight/betaflight-configurator #1892: Firmware flashing not working on MacOS with master

etracer65, 2020-02-21:
Not sure exactly when this problem started, but I remember seeing it in the past so I don't think it's been recently introduced. On MacOS after clicking the "Flash" button the board will switch to DFU mode and be detected, but flashing fails. Not errors, just "Device opened", "Device closed" messages immediately. Looking at the console I see:
```
SERIAL: Connection opened with ID: 1, Baud: 115200
MSPConnector.js:55 Connected
stm32.js:152 Looking for capabilities via MSP
stm32.js:163 no flash bootloader detected
serial.js:265 SERIAL: Connection with ID: 1 closed, Sent: 19 bytes, Received: 85 bytes
MSPConnector.js:71 Disconnected
port_handler.js:32 PortHandler - Removed: /dev/cu.usbmodem221,/dev/tty.usbmodem221
stm32usbdfu.js:95 USB DFU detected with ID: 2
stm32usbdfu.js:134 Device opened with Handle ID: 7
stm32usbdfu.js:108 reporting chrome error: Error claiming interface.
stm32usbdfu.js:160 Failed to claim USB device!
stm32usbdfu.js:1080 Script finished after: 0.012 seconds
stm32usbdfu.js:164 Claimed interface: 0
stm32usbdfu.js:178 Released interface: 0
main.html#progressbar:1 Unchecked runtime.lastError: Error releasing interface.
stm32usbdfu.js:149 Device closed with Handle ID: 7
stm32usbdfu.js:108 reporting chrome error: No such connection.
main.html#progressbar:1 Error handling response: TypeError: Cannot read property 'resultCode' of undefined
    at chrome-extension://ajdoecjhplhoahggfmgflfnlechcljoc/js/protocols/stm32usbdfu.js:274:72
```

with the relevant errors I chink being:
```
stm32usbdfu.js:108 reporting chrome error: Error claiming interface.
stm32usbdfu.js:160 Failed to claim USB device!
```

I'm not 100% sure but I thought that we had seen something similar in the past that turned out to be related to bugs in a NW.js version and needing to use a later.

Also probably related: Clicking the "Exit DFU Mode" button doesn't work and generates the same errors.

mikeller, 2020-02-22:
Yeah, we have had these in the past, and they needed an update of the NW.js version.

Can you please check if #1893 fixes this?

etracer65, 2020-03-10:
I tried some things but couldn't find any solutions. Tried the latest stable 0.44.4 and even the current beta 0.45.0-beta1 and still the same error when trying to claim the interface.

Possibly it would be fixed with node 13 but that's beyond my experience to test out.

The only solution I see at this point is to comment out the error check for MacOS as it seems to be a false error. Commenting out the error check seems to allow it to work reliably. I guess the worst case is that claiming the interface might **really** fail and I assume the subsequent flash attempt would fail so I'm not sure there's any harm in that. Not sure how to cause a legitimate case of of the interface claiming to fail to test. And and if it does in rare cases it's certainly better that always failing as it does now.

mikeller, 2020-03-11:
Thanks for testing @etracer65. We can probably skip the error check only for MacOS to minimise the potential impact. Do you want to propose a solution?

etracer65, 2020-03-11:
Yes, I’ll work up a PR.

PR #1914 (2020-03-13): Work around false error when claiming interface to fix firmware flashing on MacOS
Fixes #1892 

The error check after attempting to claim the USB interface was falsely indicating failure on MacOS even though it was working properly. Tried all available versions of NW.js (including beta) and none resolve the problem. The problem only seems to affect MacOS.

The error check after claiming the interface is suppressed for MacOS to work around the problem. Risk is low because even if the climing of the interface failed the worst that could happen is flash fails anyway.

Hopefully there will be some [... cut]
closed 2020-03-13 by mikeller