betaflight/betaflight-configurator #1899: Firmware flashing stopped working

c3n, 2020-02-26:
I can't update firmware through betaflight-configurator anymore. It used to work fine (Windows 10), now it doesn't.
- When I click "flash firmware" it restarts the FC into DFU mode and then seems to restart it again right back out of it. The bar is then waiting forever for the FC to enter bootloader.
- When I enter DFU mode manually or through button  the GUI shows DFU mode at the top right right until I click flash firmware (without the reboot procedure) and it immediately reboots the FC back into normal mode, too
- Using dfu-util instead of betaflight-configurator and a hex file converted to bin I can successfully flash the FC

Unfortunately I have no idea on how to enable further debugging notices besides the normal log at the top of the screen which just says "detected 512Kib controller"

Any ideas?

mikeller, 2020-02-26:
@c3n: What version of configurator are you using?

c3n, 2020-02-26:
I had a snapshot 10.7 but downgraded to 10.6 to make sure thats not the problem

mikeller, 2020-02-26:
So this is happening with 10.6.0 as well? Looks like it is a problem on your machine then, 10.6.0 has been out for months, and nobody else has reported this.

c3n, 2020-02-26:
Yes, I think so too. But is there a way to enable verbose logging? Because running dfu-util on the command line to flash betaflight works flawlessly

mikeller, 2020-02-26:
@c3n: You'd have to build and run the debug version by checking out locally and then running it with `yarn start` - then you will be able to open the debug console.

c3n, 2020-02-26:
Ok thanks

closed 2020-02-26 by c3n