betaflight/betaflight-configurator #1919: After paste diff all in CLI, one OSD element does not show in BF Configurator

0crap, 2020-03-15:
Using the latest BF configurator nightly build I found an issue.

When I flash my board with a full chip erase and paste in my config, one element is missing.

set osd_rssi_dbm_pos = 87

It does not show on the OSD tab in the BF Configurator.

0crap, 2020-03-16:
Added info:

1. Paste diff all does not show OSD element in Configurator which is enabled in the diff all.
2. Reproduce by flash the board with a full chip erase and paste below config
'''# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Mar 16 2020 / 05:29:45 (9b6911f80) MSP API: 1.43
# manufacturer_id: MTKS   board_name: MATEKF722   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722
manufacturer_id MTKS
mcu_id 002700193237510d31313935
signature 

# master
set acc_calibration = -4,19,13,1
set osd_rssi_dbm_pos = 87
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save'''
3. Using a MATEK F722 FC and BF configurator 10.7.0 - v20200311.163
Will always reproduce until I tick the box in the configurator.
This will not alter the diff all because it's already enabled, but it is cached in the Configurator or something.

McGiverGim, 2020-03-16:
Thanks for reporting!! Fixed in the firmware PR.

closed 2020-03-17 by mikeller