betaflight/betaflight-configurator #1925: OSD Not working with BF 4.1.5 and Foxeer F722

JosepBC, 2020-03-19:
Hello, I've just installed the 4.1.5 on one of my quads, everything works fine but the osd does not work.
If I try to change the OSD font on the 4.1.5 it doesn't work neither, but if I go back to the 3.5.3 all good, the font changes correctly and the OSD works.
I have the lipo connected.

Diff of my quad:
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.5 Mar 16 2020 / 05:21:46 (d4e74e39c) MSP API: 1.42
# manufacturer_id: FOXE   board_name: FOXEERF722DUAL   custom defaults: YES

# start the command batch
batch start

board_name FOXEERF722DUAL
manufacturer_id FOXE

# name: BelloDDC

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# serial
serial 1 2 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# aux
aux 0 0 0 1825 2100 0 0
aux 1 46 3 1350 2100 0 0
aux 2 13 2 1300 2100 0 0
aux 3 35 1 1300 2100 0 0

# vtxtable
vtxtable bands 7
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 EACHINE  U FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740 5880 5917
vtxtable band 7 LOWRACE  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = CRSF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1100
set vbat_max_cell_voltage = 420
set vbat_scale = 112
set ibata_scale = 276
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set pid_process_denom = 1
set osd_cap_alarm = 1100
set osd_rssi_pos = 2167
set osd_link_quality_pos = 2103
set osd_rssi_dbm_pos = 87
set osd_tim_1_pos = 2467
set osd_tim_2_pos = 2435
set osd_vtx_channel_pos = 2069
set osd_current_pos = 2516
set osd_mah_drawn_pos = 2484
set osd_craft_name_pos = 2474
set osd_gps_lon_pos = 2120
set osd_gps_lat_pos = 2088
set osd_gps_sats_pos = 2082
set osd_home_dir_pos = 2413
set osd_home_dist_pos = 2443
set osd_flight_dist_pos = 395
set osd_altitude_pos = 2112
set osd_warnings_pos = 2280
set osd_avg_cell_voltage_pos = 2452
set osd_battery_usage_pos = 2056
set osd_disarmed_pos = 2314
set osd_flip_arrow_pos = 235
set osd_stat_rtc_date_time = ON
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_bbox = OFF
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658
set gyro_1_align_yaw = 2700
set gyro_2_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set name = BelloDDC

profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_expo = 40
set pitch_expo = 40
set yaw_expo = 40
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# end the command batch
batch end

JosepBC, 2020-03-19:
It starts failing on bf 4.1.3

Asizon, 2020-03-19:
@mikeller maybe we can add a message in some part of font uploader to avoid this cases when battery needed before plugin usb

etracer65, 2020-03-19:
@JosepBC Start with resetting to defaults (`defaults` command in the CLI).

Then connect the battery **before** connecting the USB (PROPS REMOVED!).

mikeller, 2020-03-19:
@Asizon: See also https://github.com/betaflight/betaflight-configurator/issues/1816#issuecomment-601246789. Do you have time to come up with something?

Asizon, 2020-03-20:
yes @mikeller im now checking all of this and thinking about it

closed 2020-04-26 by stale[bot]

paulo-serrao, 2020-05-24:
Just to confirm that connecting the lipo BEFORE connecting the USB solved the issue as @etracer65 suggested. I had already tried the other way around (lipo after USB) and that causes the FC to crash.

mikeller, 2020-05-24:
@paulo-serrao: Can you do a quick test for me with the FOXEERF722DUAL: Can you please confirm that, when you are running 4.2.0-RC2, and connecting to configurator 10.7.0-RC1:
- the 'Font Manager' button appears in the bottom right when a battery was plugged in before connecting USB;
- the 'Font Manager' button does not appear when no battery is plugged in?

paulo-serrao, 2020-05-26:
Working as described :)

Also 'Font Manager' button does not appear when battery is plugged after USB (it would cause FC to crash on 4.1)