betaflight/betaflight-configurator #1936: Racing H7 Extreme: impossible to enter Mass Storage Device Mode

biasedlogic, 2020-03-23:
1. Describe your problem;

On FC SP Racing H7 Extreme: impossible to enter Mass Storage Device Mode to download logs from the built-in flash (or SD Card for that matter).

2. Include ways to reproduce the problem;

 a. Connect to Betaflight Configurator over micro USB cable. 
 b. Enter the Blackbox tab, click on 'Activate Mass Storage Device Mode'.
 c. Make sure logging device is selected accordingly (SD Card or Flash)
 d. Error pops up: Error / This operation is not supported by your hardware.

3. Provide as much information as possible about your hardware and software, including:

a.  what hardware / software you are using; / the version of all the software used;
Configurator: 10.6.0
Firmware: BTFL 4.1.5 (tried also 4.1.0)
Target: SPRACINGH7EXTREME

b. how things are connected / wired up.

naked FC with only the micro USB connected. Nothing is soldered yet, beyond the stacking connectors for the VTX board. VTX board is *not* present (attaching it does not change behaviour, I do not have an actual VTX module on it though).

Considering the µC is connected directly to the USB bus and that it's the newest and hottest chip out there, I can't believe it's hardware incapable of working as mass-storage device. What's up?

mikeller, 2020-03-23:
@biasedlogic: Mass storage mode has not yet been implemented for H7 boards.

biasedlogic, 2020-03-23:
Thanks, I somehow missed it... bummer for the log I've got in the onboard flash, but good it can log to SD card that's removable ;-)

WingsFishes, 2020-06-15:
After several failures by trying to launch a black box recording with my H7 sp racing extrême, I realized that I have exactly the same problem as biasedlogic.
I use Configurator: 10.6.0
Firmware: BTFL 4.1.5 (tried also 4.1.0)
Target: SPRACINGH7EXTREME
Do you some news about this feature reques?

haslinghuis, 2021-06-05:
Should be solved with https://github.com/betaflight/betaflight/pull/10247

closed 2021-06-05 by haslinghuis