betaflight/betaflight-configurator #1939: Beta flight latest release states corrupt file Mac OS

p925d, 2020-03-26:
I downloaded the the latest release of Beta Flight to my 2019 MacBook Pro and upon installation a corrupt file message was displayed. I
Aborted and reloaded the file numerous times for the same message to appear. 

I have used earlier version of Betaflight on the MacBook prior to this issue. 

Please advise what the issue could be. 

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

p925d, 2020-03-26:
Guys I’m not a coder, so I don’t know what would be considered missing. I have the latest release of 
macOS Catalina 10.15.4 and am trying to load BetaFlight 10.6.0 

Below is the image of the response received when I try to open the application. 

[image]

mikeller, 2020-03-26:
Thanks @p925d. The version numbers and exact error message go a long way.

p925d, 2020-03-29:
Mike, any updates on this instill cannot get the current version to download to the MacBook. So I loaded a older version which is working and updated to the latest firmware availed for my Emax mini magnum f4 stack. Another issue popped up that 1 motor will wind up and in addition when using BLHeli to read the ESC’s the system states that the ESC’s have been read but there is no data or settings to be displayed. 

Please advise. Thanks

mikeller, 2020-03-29:
Duplicate of #1675.

closed 2020-03-29 by mikeller