betaflight/betaflight-configurator #1940: Cmd+H on macOS permanently hides Configurator window

naorunaoru, 2020-03-26:
Problem:
When I hide Configurator using cmd+H shortcut the window hides, but I can't bring it back again. Clicking on the icon in the dock does not help, only restart helps.

Steps to reproduce:
- open Configurator
- press cmd+H
- try to bring back the window.

macOS 10.15.2, Configurator 10.6.0.

mikeller, 2020-03-27:
@etracer65: Can you check this please?

etracer65, 2020-03-29:
I can confirm this problem. It's not dependent on the MacOS version as it happens on 10.13.x as well. Also tested with master 10.7 and it still has the problem.

Don't know of any workarounds and seems to be a Nw.js problem. It behaves strangely when hiding the application. Every other app will hide and switch to the next app. But the configurator hides the windows but remains the active app. Then when trying to reactivate (unhide) it MacOS seems to think there are no windows associated with the app.

etracer65, 2020-03-29:
And it's not a new problem as it exists in the 10.5 version as well.

etracer65, 2020-03-29:
I don't know where it should go, but it seems like we need a `win.show()` and `win.focus()` when the app is brought to the foreground.

mikeller, 2020-03-29:
@etracer65: Could be here: http://docs.nwjs.io/en/latest/References/Window/#event-restore
Do you have time to experiment with this? I do not currently have access to a Mac.

mikeller, 2020-04-29:
Probably worth testing with 10.7.0-RC1 - we have changed the type of the main window from chrome to NW, this might make a difference.

robertlacroix, 2020-08-31:
> Probably worth testing with 10.7.0-RC1 - we have changed the type of the main window from chrome to NW, this might make a difference.

@mikeller still a problem in `10.7`

LekeFly, 2021-02-27:
This is still a problem in Betaflight Configurator 10.8 and Betaflight Blackbox Viewer nightly. A very annoying bug.

haslinghuis, 2021-03-14:
I think this setting in `package.json` will have some influence:

```
  "chromium-args": "--disable-features=nw2",
```

mikeller, 2021-03-15:
@LekeFly: Can you please check if the problem still exists when using the latest development version of configurator from https://github.com/betaflight/betaflight-configurator-nightlies/releases?

ctzsnooze, 2021-03-16:
This is still an issue in blackbox explorer nightly, for sure, and has been for a while.
Configurator behaves exactly the same way.
Usually on a Mac selecting Hide or Cmd-H will cause the app's name to 'disappear' from the top left, and all windows to 'disappear'.  To bring it back, it can be selected in the dock or via going cmd-tab and selecting it.  On selecting it, the app takes over the menu bar and the windows reappear.
In both Configurator and BBE, selecting cmd-H causes the window to disappear, and the name of the app to flash, but it stays at the top of the screen, as if it hasn't hidden.  Attempting to open the window using the Betaflight Configurator > Window menu doesn't work.  Nor does switching to another app and then back to Configurator.  Additionally, Configurator is now unresponsive and cannot be quit even if it is the frontmost app.  
The only way to fix it is to force quit and re-open.

PR #2446 (2021-03-21): Fix Mac OSX Hide command problem
Fixes: #1940 by replacing `hide` window on MacOS with `minimize`. Only quirk with this solution that `quit` does not work in minimized state but it's a tradeoff with the `hide` function making `restore` imposible and having to force quit.

Thanks @ctzsnooze for testing. Tried to remove `"chromium-args": "--disable-features=nw2"` but this is still needed for file operations, opening links and changing settings as dark mode. Thanks @mikeller for pointing that out on Slack. See #1982
closed 2021-03-21 by mikeller