betaflight/betaflight-configurator #1954: Hobbywing Xrotor F4 G3: Mass Storage Device Mode is not working on Mac OS

bodja, 2020-04-10:
#### 1. Describe your problem;
Cannot activate mass storage device mode on mac os. **(on Windows works fine)**

#### 2. Include ways to reproduce the problem;

1. Format SD card to fat32.
2. Insert SD card into a flight controller.
3. Open "Betaflight-configurator" app.
4. Connect to the flight controller.
5. Navigate to the "Blackbox" tab.
6. Choose SD Card as logging device. Chose rate and mode.
7. "Save and reboot"
8. Connect the FC again
9. Go to the "Blackbox tab"
10. Click on the "Activate Mass Storage Device Mode" button.

Expected:
Disk is successfully mounted and accessible for usage.

Actual:
Disk partition is never mounted.

I've tried basic commands to mount.
```
$ diskutil list
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *31.9 GB    disk2
   1:                 DOS_FAT_32 BLACKBOX                31.9 GB    disk2s1

$ diskutil mount /dev/disk2s1
Volume on disk2s1 timed out waiting to mount

$ sudo mkdir /Volumes/BLACKBOX
$ sudo mount -w -t msdos /dev/disk2s1 /Volumes/BLACKBOX
mount_msdos: /dev/disk2s1: Resource busy
mount: /Volumes/BLACKBOX failed with 71
```

### 3. Software/hardware info
**Configurator**: 10.6.0
**Firmware**: BTFL 4.1.4
**Target**: AIRB/OMNIBUSF4SD(STM32F405)
**FC**: Hobbywing Xrotor F4 G3
**SD card**: SanDisk Extreme 32GB

Docteh, 2020-04-10:
Hi, 
Have you tried asking the flight controller to go to Mass Storage mode by entering `msc` into the CLI?
I think this issue belongs as part of `betaflight`, not `betaflight-configurator`.

We can just transfer the issue over if that is the case.

bodja, 2020-04-10:
@Docteh Hi, thanks for the quick response. Yes `msc` command results into the same issue. Will you transfer the issue or should I do it myself?