betaflight/betaflight-configurator #1958: Can’t choose target devices nor the firmware in firmware flasher

ongeschikt88, 2020-04-11:
On MacOS  I can’t flash firmware for the flight controller. In the tab firmware flasher there are no targets in the dropdown box. 

It happens every time when I try to flash firmware. I use the latest betaflight configurator. Even reinstalled betaflight but still doesn’t work.

What I expected to happen was the same as you can find in al videos you can find online

I have a mamba f722 flight controller

jflyper, 2020-04-11:
Sounds like the configurator is not granted for Internet access; it keeps saying loading because it's continuously trying to fetch firmware metadata from GitHub which it can't.

ongeschikt88, 2020-04-11:
Problem solved. Betaflight didn’t had access to internet. Thanks for the help!

Nico

closed 2020-05-20 by stale[bot]