betaflight/betaflight-configurator #1961: broken configurator tabs

TheIsotopes, 2020-04-12:
@mikeller  since #1956 a few tabs are no longer displayed correctly

I found this on the following tabs receiver, motors and blackbox

[image]

[image]

[image]

mikeller, 2020-04-12:
Uaaa, this CSS is an abomination wrapped into a nightmare. :sob:

McGiverGim, 2020-04-12:
True, our css is a mesh... But a complete redo is frightening 😱

mikeller, 2020-04-13:
@McGiverGim: If a complete redo, then it definitely should not be done with hand-crafted styling in the first place. But agreed, a complete redo will include a scary amount of work to get to the level of completeness we are at now.

PR #1963 (2020-04-20): Fix the landing page - again. And fix the tabs broken in the last fix.
Fixes #1961.

Thanks for @TheIsotopes for pointing this out in https://github.com/betaflight/betaflight-configurator/issues/1961#issue-598440703

[image]
closed 2020-04-20 by mikeller