betaflight/betaflight-configurator #1972: dfuDNBUSY reported when erasing flash on STM32H743VIT6

mikeller, 2020-04-18:
Reproduction:

- connect a board with STM32H743VIT6 MCU in DFU mode;
- select a target for this board;
- enable 'Full chip erase';
- click 'Flash Firmware'.

Expected behaviour:

- the flash is erased without errors being reported

Actual behaviour:

- errors are shown in the console:

```
Address load for option bytes sector succeeded.
stm32usbdfu.js:702 Option bytes read successfully
stm32usbdfu.js:703 Chip does not appear read protected
stm32usbdfu.js:809 Executing local chip erase Array(16)0: {sector: 0, page: 0}1: {sector: 0, page: 1}2: {sector: 0, page: 2}3: {sector: 0, page: 3}4: {sector: 0, page: 4}5: {sector: 0, page: 5}6: {sector: 0, page: 6}7: {sector: 0, page: 7}8: {sector: 0, page: 8}9: {sector: 0, page: 9}10: {sector: 0, page: 10}11: {sector: 0, page: 11}12: {sector: 0, page: 12}13: {sector: 0, page: 13}14: {sector: 0, page: 14}15: {sector: 0, page: 15}length: 16__proto__: Array(0)
stm32usbdfu.js:832 Erasing. sector 0, page 0 @ 0x8000000
stm32usbdfu.js:832 Erasing. sector 0, page 1 @ 0x8020000
stm32usbdfu.js:832 Erasing. sector 0, page 2 @ 0x8040000
stm32usbdfu.js:832 Erasing. sector 0, page 3 @ 0x8060000
stm32usbdfu.js:832 Erasing. sector 0, page 4 @ 0x8080000
stm32usbdfu.js:832 Erasing. sector 0, page 5 @ 0x80a0000
stm32usbdfu.js:832 Erasing. sector 0, page 6 @ 0x80c0000
stm32usbdfu.js:832 Erasing. sector 0, page 7 @ 0x80e0000
stm32usbdfu.js:832 Erasing. sector 0, page 8 @ 0x8100000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 9 @ 0x8120000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 10 @ 0x8140000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 11 @ 0x8160000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 12 @ 0x8180000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 13 @ 0x81a0000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 14 @ 0x81c0000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:832 Erasing. sector 0, page 15 @ 0x81e0000
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
stm32usbdfu.js:819 Erase: complete
```

STM32H743VIT6 is listed as a 2MB part, but it looks like trying to erase the second MB results in timeouts.

jflyper, 2020-05-18:
I think I've encountered this problem before ...

jflyper, 2020-05-18:
What is the silicon revision of your chip?

mikeller, 2020-05-18:
@jflyper:

`MCU H743 (Rev.V)`

Are you thinking of #1861?

jflyper, 2020-05-18:
Ah, yes. I didn't remember the exact number.

I think this issue has been worked around by #1861 configurator-wise.
The
```
stm32usbdfu.js:854 erase_page: dfuDNBUSY after timeout, clearing
```
log messages are expected.

What I don't remember/wonder is if the erase was actually successful or not...

mikeller, 2020-05-19:
@jflyper: Heh yeah, looking at the code of #1861 this seems to be intentional - closing.

closed 2020-05-19 by mikeller