betaflight/betaflight-configurator #1974: Video transmitter tab broken

cschneider, 2020-04-19:
I connect to the drone, then switch to Video transmitter tab.

[image]

There seem to be some UI items missing. 
Most importantly the buttons on the bottom right are only shown as yellow lines with no text. Clicking them does not seem to have any effect.

Once I have this tab I can not click any other tabs. The only way to get the configurator working again is to disconnect and connect again.
The tab used to work for me. So maybe this error only appears in combination with certain current values in the vtx.

I tried to list the current vtx values from the CLI. Not sure if they help:
# get vtx
osd_vtx_channel_pos = 234
Allowed range: 0 - 15359

vtx_band = 1
Allowed range: 0 - 8
Default value: 0

vtx_channel = 0
Allowed range: 0 - 8

vtx_power = 2
Allowed range: 0 - 7
Default value: 0

vtx_low_power_disarm = OFF
Allowed values: OFF, ON, UNTIL_FIRST_ARM

vtx_freq = 0
Allowed range: 0 - 5999

vtx_pit_mode_freq = 0
Allowed range: 0 - 5999

vtx_halfduplex = ON
Allowed values: OFF, ON

Asizon, 2020-04-19:
Fixed in https://github.com/betaflight/betaflight-configurator/pull/1849

Can you please try Configurator 10.7.0 to confirm?you can download here https://github.com/betaflight/betaflight-configurator-nightlies/releases

Please come back to 10.6.0 when you tried it. 10.7.0 is in development.

cschneider, 2020-04-19:
I tried v20200413.352 but with this version I can not even connect to the drone. I see the usbmodem tty but when I connect to it, it simply times out.

cschneider, 2020-04-19:
I now tried v20200412.346. With this version the connect is working.

cschneider, 2020-04-19:
FYI I now first tried to install the vtx table for smartaudio EU. Loading this table also leaves the UI in an unusable state. Then I loaded the US table and it worked.

closed 2020-05-30 by stale[bot]