betaflight/betaflight-configurator #1975: 10.6 Won't install on MAC OS Catalina - 10.15.4 (19E266)

tahoematt, 2020-04-19:
I use betaflight on my Macbook pro. It is running 10.15.4 (19E266) - 10.5 works fine but I want to run 10.6 so I can use 4.1.

When I try to install Betaflight 10.6 I get the following error:
"Betaflight Configurator.app” is damaged and can’t be opened. You should move it to the Trash.
betaflight-configurator downloaded this file today at 3:33 PM from github.com.

I ran the following suggested terminal code before installing it. 

sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app

Nothing works. Please help.

Marlowe17, 2020-04-22:
im agreed with you! same problem with my MacBook Pro after updated 10.15.4

etracer65, 2020-05-03:
Works fine on MacOs catalina.

Download the .dmg file. Double click to open it. Drag the Betaflight Configurator icon to the Applications folder. Open a terminal and paste in the:
```
sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app
```
You'll need to enter your administration password. If the command returns any errors either your password wasn't correct, you're not an admin user, or you made a mistake with the command.

xrshck, 2020-05-04:
Works fine for me too on Catalina Beta 10.15.5.

tahoematt, 2020-05-04:
Got it to work

mikeller, 2020-05-04:
@tahoematt: What was the trick?

GuidoHS, 2020-05-05:
The information is confusing on the installation notes. You should run the line of code after installing the app not before. If you don't already have an earlier  version installed you will also get an error code because the configurator does not exist on your system yet. Alternatively you can also go into system preferences and approve the app manually without the terminal code.

mikeller, 2020-05-06:
@GuidoHS: Not really sure what is confusing about this:

> To work around this, run the following command in a terminal after installing: `sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app`.

Can you make a suggestion on how this could be made more clear?

GuidoHS, 2020-05-07:
The fact that it says you have to run the code before installing and not after:

> Changes to the security model used in the latest versions of MacOS X 10.14 (Mojave) and 10.15 (Catalina) mean that the operating system will show an error message ('"Betaflight Configurator.app" is damaged and can’t be opened. You should move it to the Trash.') when trying to install the application. **To work around this, run the following command in a terminal before installing:** sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Configurator.app.

GuidoHS, 2020-05-07:
But I guess it's safer and better to recommend everyone to ignore the message. Then go to system preferences > security >(  unlock if needed >) click the button that's there saying to open this app anyways.

This might require adjusting another setting in that same menu tab but I'm writing this from memory on my phone atm.

McGiverGim, 2020-05-07:
In the Readme of the repo it says "after", but the release notes say "before". 😮

mikeller, 2020-05-07:
@GuidoHS: 

> To work around this, run the following command in a terminal before installing

Where did you find this text? As @McGiverGim pointed out the README says _after_, but there might be an old version around somewhere that is wrong.

GuidoHS, 2020-05-07:
At the 10.7 release, released by you 9 days ago?

https://github.com/betaflight/betaflight-configurator/releases

mikeller, 2020-05-07:
@GuidoHS: Lol, I copy pasted that one from https://github.com/betaflight/betaflight-configurator/releases/tag/10.6.0 (okok, was still me initially adding it), and nobody ever complained about it in 10.6.0.

Thanks for letting us know, fixed now.

GuidoHS, 2020-05-07:
No worries. It's not a problem if you understand the error message or use the native button approval but if you have no clue and just follow instructions to the detail you're stuck :p 
I only ran into it because I'm having the save and reboot issue and therefore tried reinstalling exactly as per the instructions.

closed 2020-06-13 by stale[bot]