betaflight/betaflight-configurator #1980: Hiperlinks are broken

Asizon, 2020-04-23:
Href attribute is broken and unable to load the requested links. This occurs throughout all of the Configurator Hiperlinks:

[image]

I can see that removing `ref: noopener nooreferrer` it works as expected, but idont know if remove al refs is the right solution:

[image]

McGiverGim, 2020-04-23:
Maybe I'm wrong, but I think this is only a secondary effect of the new-win-policy not executing.

https://github.com/betaflight/betaflight-configurator/blob/f9235b54f70ad4103737b28f1a568437694deaf5/src/js/main.js#L87-L103

Doing a debug, the `windows` variable seems empty, so it is not executing the `close` or the `new-window-policy` listeners.

McGiverGim, 2020-04-23:
It seems the open in external browser policy is broken since some versions ago... I have tested version 10.3 and it works perfectly, with or without the `noopener noreferrer` opening the link in a new browser.
With actual master or 10.5 it is not being opened in a external browser, it opens in a new window but inside de application, and this seems to not like the new links.

Docteh, 2020-04-23:
I thought it was working. I know its been wrestled with a few times.

Docteh, 2020-04-24:
Links open properly in my install of 10.6.0, 

I looked into this a bit and I'm pretty sure this is related to NW.js changing how they interact with windows https://nwjs.io/blog/nw2-mode/

https://github.com/nwjs/nw.js/issues/7377 mentions that we can temporarily disable NW2 for now, and I've filed https://github.com/nwjs/nw.js/issues/7451 

@mikeller can you confirm either failure or success on opening links on your end?

Asizon, 2020-04-24:
This is a new change introduced here:
https://github.com/betaflight/betaflight-configurator/pull/1956

McGiverGim, 2020-04-24:
Yes, I have been testing things thinking the same. 10.6 works. Executing 10.7 with `--disable-features=nw2` does not work neither.

Maybe has changed when the window is available. I will look back to see what is the latest version working of nw.js.

We can fix it too with jquery intercepting the external link calls, but I don't know if it is the best solution...

Docteh, 2020-04-24:
Try reverting the code, the last change there to iterate through getAll is
where it stops working with nw1

McGiverGim, 2020-04-24:
I have used the simple get() that we have before.

Tested:
- 0.42.4 NOT WORKING
- 0.42.3 WORKING 

I will look at the changelog, but I doubt I can see something.

McGiverGim, 2020-04-24:
Is the first version with NW2 mode... so you are right @Docteh

McGiverGim, 2020-04-24:
I think we maybe use the windows in a different manner of the rest of people. Here is the issue that made the `getAll()` method appear, and it seems to work for the other user: https://github.com/nwjs/nw.js/issues/7227

So maybe our use of nw.js is a little strange...

McGiverGim, 2020-04-24:
I think maybe I know what happens... 
Our app is divided into two: 
- A background page (js/eventPage.js)
- Our "real" app (that starts in main.html)

According to node documentation, the background page does not have a corresponding window, because is a "batch" execution.

In our code we CLOSE the NW window (file main_nwjs.html) and we open the main.html from the background page. Maybe since nw2 the context of this open window remains without associated window?

I suppose this was done in this way to maintain compatibility with Chrome OS. 

I'm not too sure, but this maybe explains why this bug seems only affect to us.

McGiverGim, 2020-04-24:
I've been playing with this with different results.

After some "patches" to ignore the Chrome part and open directly the nw window, I can attach the new function for the new window policy, but I get this error after the first click on a link:
```
(BLESSED_EXTENSION context for ajdoecjhplhoahggfmgflfnlechcljoc) extensions::nw.Window:718: Uncaught TypeError: Cannot read property 'onNewWinPolicy' of null{TypeError: Cannot read property 'onNewWinPolicy' of null
    at dispatchEventIfExists (extensions::nw.Window:718:21)
    at onNewWinPolicy (extensions::nw.Window:727:3)}
```

Someone knows how to fix that? Maybe some permissions missing in the manifest?

The most curious thing is that if after that I execute this in the console: `nw.Window.get()` the opening in external browser start to work...

mikeller, 2020-04-24:
Interesting - I have been testing with hyperlinks both on 10.6.0 and `master` recently when working on the landing page, and they seem to be working fine on both on linux. Can I let you guys do the fixing / testing here, as I am not super well set up to test in Windows? Happy to test / verify particular things if you let me know.

Docteh, 2020-04-25:
We can probably handle it. I'm lost in the weeds trying to replicate success :) just tried `master` via X11 forwarding and in a virtualbox
[image]

For some reason `GUI.nwGui.Window.getAll` (or `nw.Window.getAll` comes back with an empty array. 
[image]

@McGiverGim are those changes available in a public/private branch? I've forgotten some of the git terminology, I want to pull your change down from your fork if its pushed there.

McGiverGim, 2020-04-25:
No, I will try to push it in some hours, but is only a patch to verify my supposition, is not a solution to the problem.

mikeller, 2020-04-25:
@Docteh: Sorry, that was a red herring - when testing again I found that it is broken in linux as well - I suspect the successful testing was in a pull request that came before the one where I added `rel="noopener noreferrer"` to the links on the landing page.
A couple of remarks:
- I think @McGiverGim is right, and the problem stems from our main app not having a window associated - when checking the event handler that we use to open links in an external app I found that this is not actually set for any Window:
https://github.com/betaflight/betaflight-configurator/blob/aa98588379a1651375f742e0737516b0ad011fb8/src/js/main.js#L88-L102
- if we manage to fix this (might need some jiggery pokery to pass in the window of the main thread), and the `new-win-policy` handler gets used again, then I am not sure if we need the `rel="noopener noreferrer"` since we are already opening the links in an external app, thusly preventing forcing the new page to be loaded in a new process;
- now checking if upgrading NW.js to 0.45 will fix this, otherwise I recommend downgrading to 0.43 until we have found a fix for this.

Docteh, 2020-04-25:
https://github.com/Docteh/betaflight-configurator/tree/nw2 This is what I came up with, but ports do not appear to be scanned initially, and the `onClose` stuff in `eventPage.js` is not dealt with. 

`src/js/port_handler.js` has some 10 second timeouts that might be relevant to the serial ports not showing immediately. 

`rel="noopener noreferrer"` is something sonar cloud complained about. Gotta have secure links, yo.

Since we're intending to assign the `new-win-policy` to the current window, I think we should revert back to `nw.Window.get` instead of `nw.Window.getAll`. Well, back to `GUI.nwGui.Window.get()`

With NWjs we can disable the nw2 "feature". https://nwjs.io/blog/nw2-mode/ Says that 0.42.4 is first version  with NW2, so 0.43 may have this feature? If you pick a version I can check it out. Looks like 0.43.0 to 0.43.6 are your  options. 

I'll check back in a few more hours after @McGiverGim looks at things. 🛏️

McGiverGim, 2020-04-25:
@Docteh my branch is here: https://github.com/McGiverGim/betaflight-configurator/tree/test_fix_external_url

I simply changed the event page where we open a Chrome window:
```
chrome.app.window.create('main.html', {
```

By a node_main.js page where we open a nw window:
```
nw.Window.open('main.html',
```
With this the links work because the `getAll` gets all the windows.
This is not a fix, because:
- We lost compatibility with Chrome.
- We need to add the close listener to the nw windows to close the connections, like it is done in the Chrome window. Can be added easy, but it has not been done.

So it seems clear that maybe the nw.js people have forgotten to attach the nw window at the chrome window in nw2.

@mikeller I think is a good idea to maintain the `noopener noreferrer` in the links. Maybe in a future we can publish this as a web app and any change makes the translators change it again. And the problem is not in the clausule, simply it makes the exisiting bug appear bigger.

McGiverGim, 2020-04-25:
I forgot to say: 0.45 does not fix this. I tested yesterday. The latest version that works out of the box is 0.42.3 (the latest without nw2).

Docteh, 2020-04-25:
Oh, NWjs is intentionally migrating away from the `chrome.app` interface with nw2, so we'll lose it eventually anyways.

McGiverGim, 2020-04-25:
Then I think we must release with version 0.42.3 and remove the Chrome support for next version, as we have discussed several times :(

mikeller, 2020-04-25:
#1982 should fix this. Turns out that #782 was always a hack, it would have sufficed back then to add an id for the application window to enable size / position restore.

Re `noopener noreferrer`, a couple of points:
- setting this does not do anything when opening links in an external app;
- the functionality is is already incomplete as it is, as this is not set in any dynamically loaded links (like the release notes).

To me, both of these together strongly suggest that we should not go on and add more complexity to set this on dynamically loaded links for something that does not do anything.

Re. future plans for a webapp, let's try and not build things for future eventuality - experience tells me that this is almost always a waste of time, as by the time something is used the conditions will have changed.

Also re translators, I think this is a good argument for not setting `noopener noreferrer` in the way we are doing it - it just adds to the effort required for translations. When we get to a point of needing it, we should rather look into adding this dynamically.

But pulling these out can wait until 10.7 has gone out.

Asizon, 2020-04-27:
Thankyou guys for getting involved in solving this, seems that finnaly we have a temporal solution for NW2(remove it temporaly).

PR #1982 (2020-04-28): Fixed opening of links in an external app.
Fixes #1980.

Open the app Window as a NW.js window, which is what NW.js was designed for.
This will re-enable the `new-win-policy` NW.js event handler that is required to open links in an external app.

Also moves the `close` handler into `main.js`, so that `eventPage.js` does not need to be used in NW.js any more.
closed 2020-04-28 by mikeller