betaflight/betaflight-configurator #1991: (Suggestion) Help prevent flashing the wrong device

Juzzle1, 2020-04-28:
**The problem:**

STM32 based OpenTX devices connect in USB bootloader mode when powered off.

Betaflight configurator immediately writes to the first bootloader device it sees after the user clicks flash firmware, and if an OpenTX device is plugged in as well as the flight controller, it writes to the wrong device.

While this could be described as user error and it's easy to re-flash OpenTX, it could be a problem for certain OpenTX devices (e.g. FrSky X9D) which store model data in flash, as model data could be erased.

**Proposed solution:**

When 'Flash Firmware' is clicked, check for existing bootloader devices _before_ rebooting the flight controller into bootloader mode.

If any are found, pop up a warning to suggest the user may wish to disconnect other USB devices first to prevent flashing the wrong device.

mikeller, 2020-04-29:
@Juzzle1: I see this more as a bug in OpenTX than something that should be fixed by every other app that contains code to flash STM32 MCUs - they should not just be dropping into DFU mode without the user indicating that they want to flash firmware - this does not make sense.
Point in case: The STLink utility, and `dfu-util`, and every other DFU tool out there will be doing the same when a device running OpenTX is connected while powered off - should an 'OpenTX' warning be built into all of these as well?

Juzzle1, 2020-04-29:
@mikeller 

I agree with you, however for 99% of Betaflight users, the only other STM32 device that would ever be plugged in to their PC would be an OpenTX device, and getting the OpenTX devs to change this fairly fundamental behaviour is unlikely to happen any time soon. 

My thoughts were simply that this would be a fairly clean and simple way of making Betaflight / OpenTX just a little bit more friendly especially to novice users.

On the flipside, OpenTX does have a 'Check Hardware compatibility' feature to prevent this happening the other way round.

Just my 2 cents here, thanks for all you do, Betaflight is awesome :)

mikeller, 2020-04-29:
@Juzzle1: Not sure this is a problem that needs solving: There is no point for any user to ever connect a TX running OpenTX via USB while it is powered off, as what it does in this case (drop into DFU mode) does not have any practical purpose, so it is probably safe to assume that users are not doing this.
On the other hand, there is value in the behaviour that is currently implemented in Betaflight Configurator which is to attempt to flash devices connected to it in DFU mode: This is used by users with entirely new boards that have no firmware installed, and by users recovering after a failed firmware flashing attempt (boot button).
Changing the reasonable behaviour in configurator (that is in line with every other tool out there) to accommodate the result of the incompetent design of OpenTX just does not make sense. If you want to get this addressed, open an issue with https://github.com/opentx/opentx and ask them to fix the pointless behaviour of their firmware.

Juzzle1, 2020-04-29:
@mikeller agreed it is very dumb behaviour, and I'll take it up with the OpenTX devs.

There are legitimate reasons for having devices plugged in and turned off, for example the Jumper T16 charges via USB.

Cheers

closed 2020-04-29 by Juzzle1

mikeller, 2020-04-30:
@Juzzle1:

> There are legitimate reasons for having devices plugged in and turned off, for example the Jumper T16 charges via USB.

Ok, I did not know that - even more so the behaviour of OpenTX should be changed - having a device plugged in that registers as an MCU waiting to be flashed, and then drawing a high current for charging, sounds like a recipe for problems.

Juzzle1, 2020-04-30:
@mikeller Good point about the current draw, the T16 pulls around 1500mA when charging which could cause a significant voltage drop for some setups.

I've opened an issue with OpenTX, will be interesting to see what they have to say.

Juzzle1, 2020-05-01:
@mikeller OpenTX devs can't / won't do anything about it.

https://github.com/opentx/opentx/issues/7561

I guess some people are going to end up with Betaflight on their transmitters :)

mikeller, 2020-05-02:
@Juzzle1:

> I guess some people are going to end up with Betaflight on their transmitters :)

Well, one could argue that this is an upgrade from incapably developed software to useful one. ;-)
There are no plans for Betaflight to move into the TX space anytime soon, so sadly that won't help much.

But I guess the comments in opentx/opentx#7561 are right in one point: If OpenTX drops the MCU into DFU mode when powered over USB then charging the TX by connecting it to a PC is a bad idea - the power draw will be a lot higher than what is announced by DFU mode, potentially causing overloading, problems with other USB devices, or even damage.