betaflight/betaflight-configurator #1994: Betaflight unable to show modes, OSD

AndKe, 2020-04-29:
Description:
Configurator fail to display modes (and get stuck there) , and fail to display OSD (shows nothing)

Reproduce the problem;
All you need it to try to use the latest stable or release candidate of the configurator
See used hardware/firmware/configuration is here:
[BTFL_cli_Andre_20200429_220440.txt](https://github.com/betaflight/betaflight-configurator/files/4554369/BTFL_cli_Andre_20200429_220440.txt)

Yes, I do believe that the configuration (made with 10.6.0) - is the issue, it worked before, then after last OSD change stopped working, and now is not working using latest release candidate either.

Whatever corrupted (?) some settings, you may want to add handling of that parameter.

[image]

mikeller, 2020-04-29:
@AndKe: Can you please supply the output of a `diff` in CLI?

McGiverGim, 2020-04-30:
I suspect this is the problem: https://github.com/betaflight/betaflight-configurator/issues/1335

Can you flash an official version of Betaflight to your FC?

AndKe, 2020-04-30:
Hi, I am not sure what you mean by "official" version, the 4.0.0 I used was a stable/release and I downloaded/installed it configurator at some time.
As a part of troubleshooting, I did a CLI dump ,upgraded to latest build yesterday, ran into some problems, and reverted to 4.0.0 - restoring the CLI dump - and the problem was gone.. 
Then I upgraded to current release, and now are fighting with the VTX channel configuration /smartaudio that does not work - but that's irrelevant for this issue :)

So I am unable to reproduce this issue anymore.

One thing that have always puzzled me, is that I flash certain firmware, and then are offered to load some "customization" for it, - it is unclear to me (and in the docs) when one would choose not to, or - is an upgrade ment to keep settings and such customization?   I might have cocked up something there.

McGiverGim, 2020-04-30:
This was a problem that usually happened with not final 4.0 versions (beta) that come with some FCs (or if you install manually a not final released version and you didn't update it when the final version was available).

I'm sure you had one of this. When you "reverted" to 4.0 in reality you installed an official release version that fixed the problem.

Regards!

AndKe, 2020-04-30:
My flight controller was delivered with pre 3.something on it, I flew it for five minutes, then upgraded everything to latest stable due to the famous "RC-loss"   that was long time ago, and I have been running some other stable version too before 4.0.0.   - that said, it is quite possible I cocked up something at some time, once I flashed wrong build.  So you may be onto something.
Anyway - thank you.

closed 2020-04-30 by AndKe