betaflight/betaflight-configurator #1996: BF-Conf 10.7.0-RC1: Configuration tab GPS mag_declination Allowed range: -18000 - 18000

zigomatichub, 2020-04-30:
Hello
Thanks to dump all command where I see this: 
set mag_declination = 20000
###ERROR: dump: CORRUPTED CONFIG: mag_declination = 20000###

1. 
In BF-Conf , under configuration tab -> GPS 
Can you align the range to min max corresponding to the CLI command ?

2. 
In French: it's mentionned "Inclinaison du magnétomètre [deg]" but the correct translation is déclinaison magnétique [deg]

Thank you

McGiverGim, 2020-04-30:
Maybe we need to remove this parameter. Is not used in the firmware for nothing, is ignored.
Did you assign some value to it? Did you flash with erase config? You don't explain exactly what is your problem...

zigomatichub, 2020-04-30:
To reproduce the problem:
Set 20,00 into magnetometer declination into configuration tab - GPS area

go into CLI: get mag_declination

I introduce now the correct value of 1,21 deg , it's now 1210 in CLI.
So I don't have any problem. If it's not use then removed from CLI and GPS area ;)

PR #1997 (2020-05-03): Removed unused compass declination setting.
Fixes #1996.
closed 2020-05-03 by mikeller