betaflight/betaflight-configurator #2002: Anti-gravity section on PID tuning tab incorrectly hiding when feature is disabled

etracer65, 2020-05-04:
Turning the ANTI_GRAVITY feature off on the Configuration tab doesn't disable the function. Rather it causes a mode to appear on the Modes tab to allow it to be controlled with a switch (similar to how AIRMODE works).

So hiding the anti-gravity section on the PID Tuning tab based on whether the feature is enabled is incorrect. Should not ever be hidden.

etracer65, 2020-05-05:
As an alternative, the anti-gravity panel can be hidden (or better, disabled) if the feature is disabled **AND** there is not an ANTI GRAVITY mode configured on the Modes tab. But I'm not sure it's worth this level of effort.

mikeller, 2020-05-05:
I think a better way is to include the 'feature switch' for this on the panel - having the configuration for this functionality spread over 3 tabs is making it hard for users to understand what is going on.

etracer65, 2020-05-05:
But since the feature switch can’t turn it off, you’ll need both the feature switch and the “turn off” switch. Not sure how to represent that without making it more confusing.

mikeller, 2020-05-05:
@etracer65: Agreed - not sure there is a good way to make this non-confusing. But I can have a stab at it, and we can see how the result looks like.

PR #2042 (2020-06-02): Fixed the antigravity feature switch.
Fixes #2002.

[image]

[image]
closed 2020-06-02 by mikeller