betaflight/betaflight-configurator #2006: Cannot save and reboot

GuidoHS, 2020-05-05:
I cannot use the save and reboot button on the latest configurator and BF on MacOS. It simply does not respond at all rendering me unable to even save a motor protocol and so I cannot fly at all. After clicking the button the configurator freezes and I cannot access any other tabs until I disconnect and reconnect. Saving any other tab without reboot works fine. I tried reflashing a couple of times but only reflashing previous versions solves the problem. Both the previous configurator with the latest BF and the previous BF with the latest configurator actually do work. But then I cannot use the latest functionalities.

I connect via a micro USB cable. 

Extra info:
AIKON F7 v2 (DJI HD version)
BF4.2.0RC
configurator 10.7RC1
MacOS Catalina

How to reproduce:
Just flash 4.2 and try save and reboot on any of the tabs that require it.

droneXcursion, 2020-05-05:
yep. I am having the same problem. please someone fix. thanks.

McGiverGim, 2020-05-06:
@droneXcursion do you have a Mac too?

droneXcursion, 2020-05-06:
yes. this issue is definitely present on macs.

McGiverGim, 2020-05-06:
Can you give more details? mac version, fc, etc

ghden, 2020-05-06:
Same problem here with Mac:

1. Change something in configuration tab
2. Click save and reboot
3. Observe nothing happens, and many time you click on disconnect and after several attempts it responds and disconnect, and never can connect back until I unplug and plug back the FC..
4. Version 10.6, no issue at all, same firmware 4.2 RC1
Configurator: 10.7.0-RC1

ConnectUpdate Firmware 
Auto-Connect
Hide Log
2020-05-06 @ 17:43:38 -- Running - OS: MacOS, Chrome: 80.0.3987.116, Configurator: 10.7.0-RC1

2020-05-06 @ 17:43:38 -- Using cached release information for configurator releases.

2020-05-06 @ 17:43:40 -- Serial port successfully opened with ID: 1

2020-05-06 @ 17:43:40 -- MultiWii API version: 1.43.0

2020-05-06 @ 17:43:40 -- Flight controller info, identifier: BTFL, version: 4.2.0

2020-05-06 @ 17:43:40 -- Running firmware released on: Apr 29 2020 08:09:54

2020-05-06 @ 17:43:40 -- Board: IFRC/IFLIGHT_F722_TWING(STM32F7X2), version: 0

2020-05-06 @ 17:43:40 -- Unique device ID: 0x3a00223237510331313935

2020-05-06 @ 17:43:40 -- Craft name:

2020-05-06 @ 17:43:40 -- Arming Disabled

2020-05-06 @ 17:44:11 -- Serial port successfully closed

2020-05-06 @ 17:44:15 -- Serial port successfully opened with ID: 2

2020-05-06 @ 17:44:25 -- No configuration received within 10 seconds, communication failed

2020-05-06 @ 17:44:25 -- Serial port successfully closed

ghden, 2020-05-06:
CLI Save works fine..

etracer65, 2020-05-07:
I've been able to replicate this and I'm looking into it. So far it only seems to be affecting AikonF7 flight controllers with MacOS. At this point it's a connection that doesn't make much sense.

GuidoHS, 2020-05-07:
His is an iflight board though:

> 2020-05-06 @ 17:43:40 -- Board: IFRC/IFLIGHT_F722_TWING(STM32F7X2), version: 0

droneXcursion, 2020-05-07:
it's happening on my matek F7 FC too.

etracer65, 2020-05-07:
@droneXcursion Please be specific. Exactly which of the many Matek F7 types do you have?

droneXcursion, 2020-05-07:
matek f722se confirmed having this issue.

McGiverGim, 2020-05-07:
I have the same FC and under windows it seems to work.

droneXcursion, 2020-05-07:
correct. the issue is only happening on macs.

bernii, 2020-05-08:
same here, works on Win10 but does not work on MacOS Mojave/Catalina

borntocrash, 2020-05-08:
J’ai également le même problème sous Mac OS catalina avec carte Dalrc f722dual, impossible de sauvegarder et redémarrer

etracer65, 2020-05-08:
It's not necessary to post any more "me too" comments. We know what the problem is and are working on a solution.

etracer65, 2020-05-09:
I have a fix for this problem that I'd like people to test. It's a problem in the firmware - not the Configurator. Please flash this firmware and test that the Configurator now behaves normally. The firmware file is only appropriate for F722 based flight controllers (like the AikonF7).

[betaflight_4.2.0_STM32F7X2_af17ef2.hex.zip](https://github.com/betaflight/betaflight-configurator/files/4604874/betaflight_4.2.0_STM32F7X2_af17ef2.hex.zip)

If you don't know how to flash a development version of a unified target, please watch this video:

https://youtu.be/I1uN9CN30gw

bernii, 2020-05-10:
I can confirm that the above firmware fixed the problem for me (TMotor F7 with MacOS Mojave). Thank you!

closed 2020-05-13 by mikeller