betaflight/betaflight-configurator #2009: PID's and FF values erratic text cursor behaviour when typing changes manually

podgeme, 2020-05-07:
this issue seems to persist ONLY in the PID's / FF menu (that is all of the P, I , D-max, D-min and FF columns). I have checked through basically all of the other sections in the configurator where you can type in numerical values and it doesn't occur anywhere else.

[image]

The issue is that if you attempt to edit the numbers in any of these boxes by placing the text cursor anywhere except for at the end of the number then after carrying out the commanded amendment (be that delete, backspace, or typing a number) the cursor will then jump to the end of the string of numbers.

for example if you had the number "1234" in one of the boxes and you placed the text cursor between the "1" and the "2" and backspaced the "1" to delete it the text cursor will then jump to the end of the string of numbers after deleting the "1". this can lead to typing in the unintended number if you don't realise that the text cursor has moved which is how i found out. note this issue is not only present when deleting numbers but also when adding numbers to the string.

I am running Windows 10 on BF configurator 10.7.0-RC1
I have just down-dated my configurator to 10.6.0 and the issue is/was still present on that version too however it is gone on 10.5.1 . It seems that this issue has been around ever since the PID sliders were introduced and the rates broken out onto a new tab.

Asizon, 2020-05-07:
Ues, this is an important change on 10.7.0 and 4.2, if you are using 4.2,please not copypaste diffs from previous versions. What Betaflight version are you trying?

McGiverGim, 2020-05-15:
The problem is attached to the refresh of the sliders.

https://github.com/betaflight/betaflight-configurator/blob/43ea9610a67f9c40926defff4b5d4c0eb9991068/src/js/tabs/pid_tuning.js#L1847-L1850

I'm not too sure why, I will try to find the reason but I don't have too much time. Maybe others can help too starting from here.

McGiverGim, 2020-05-15:
Some more info:
That method dos some process and calls this, that changes the values of the inputs by the default PIDs:
https://github.com/betaflight/betaflight-configurator/blob/43ea9610a67f9c40926defff4b5d4c0eb9991068/src/js/TuningSliders.js#L307

And the process, later, goes here:
https://github.com/betaflight/betaflight-configurator/blob/43ea9610a67f9c40926defff4b5d4c0eb9991068/src/js/TuningSliders.js#L179
Are changed again to the values introduced by the user.
This double change is what produces the movement of the cursor.

@IvoFPV some idea what is this done in this way? Is easy to fix? I can't look at the code more at this moment.

haslinghuis, 2021-06-04:
Should be fixed in https://github.com/betaflight/betaflight-configurator/pull/2483

closed 2021-06-04 by haslinghuis