betaflight/betaflight-configurator #2016: The firmware updater not working with some fc

HatalyakMate, 2020-05-09:
I have to fc Matek F405-STD, and Furious FPV Piko f4, and the firmware updater not working. 
[image]

mikeller, 2020-05-09:
@HatalyakMate: What operating system are you using?

Also, can you try to flash using DFU mode using the sequence of steps as described in `Recovery / Lost communication` on the firmware flasher tab - please do report if the port drop-down shows 'DFU' after the 'Release BOOT button if your FC has one' step.

HatalyakMate, 2020-05-09:
I use win10
I went through all the recovery/lost communication steps but it still doesn't work, and when i push the dfu boot button and plug in, the betaflight dont see my fc

McGiverGim, 2020-05-09:
I had a similar problem and I "fixed" it trying different usb cables and different usb ports. Have you tried this?

HatalyakMate, 2020-05-09:
Yeah, i try all usb port, and all usb cable that i find, but i have another fc and this is work perfect, the matek and furious fc has this problem

mikeller, 2020-05-09:
@HatalyakMate: What model is the flight controller that is working? On the builds that do not work, do you have any RX or other components that are powered when you are connected to USB and try to flash?

HatalyakMate, 2020-05-09:
Clracing f4 is work perfect, when i try, simple fc, no rx, no solder, no anything

HatalyakMate, 2020-05-09:
Or mamba mk2 is work perfect

mikeller, 2020-05-10:
This is super weird - it looks like some of your boards are recognised when in DFU mode, others aren't. Can you please test with one board that isn't recognised in DFU mode: Put it into DFU mode (push Boot button, connect USB, release Boot button). Then open device manager in Windows, and check if there are any errors detected, or otherwise search for the DFU device and provide a screenshot of this.

MATEKSYS, 2020-05-10:
This issue should be related to DFU driver. the DFU driver worked with dfusedemo will not work with BF/INAV configurator.  and vice versa.
I used to use dfusedemo occasionally,  I have to uninstall DFU driver which works with configurator first, then restall DFU driver for dfusedemo.

MATEKSYS, 2020-05-10:
DFU driver working with dfuse is "STM Device in DFU mode"
[image]

After replacing drivers with ImpulseRC_Driver_Fixer or zadig , the device will change to "STM32 BOOTLOADER", this device will not be detected by dfusedemo
[image]

mikeller, 2020-05-10:
Thanks @MATEKSYS. I am not a Windows user so I am not familiar with 'DfuSe Demo', and did not know that this does not use the official STM DFU driver. But this would be a reason for why this is not working. It's just surprising that it still seems to be working for some boards - could be that the DFU interface registers with a different ID for different MCU types.

mikeller, 2020-05-10:
@HatalyakMate: Can you please install the correct drivers using the sequence of steps as described in 'Recovery / Lost communication' on the firmware flasher tab, and then report back if this fixes it for you?

HatalyakMate, 2020-05-10:
Okey, i try the zadig but zadig cant see my fc, (normal and dfu mode) but the iflight driver fix is working, fix my driver, but on the betaflight nothinh changed

HatalyakMate, 2020-05-10:
so, zadig is find my fc, fixed driver, but now, betaflight cant see, and i got a yellow triangle on device manager

HatalyakMate, 2020-05-10:
and finaly, i run again the impulse rc driver fix, and now, when i connect my fc in dfu mode, the betaflight see, and the flashing is working
[image]

HatalyakMate, 2020-05-11:
Thanks for help everybody ✌️

closed 2020-05-11 by HatalyakMate