betaflight/betaflight-configurator #2019: Betafligtht opens in wrong language

C4Terminator, 2020-05-11:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

C4Terminator, 2020-05-11:
Betaflight Configurator changed languages on me. It should be English. how do i fix this???

etracer65, 2020-05-11:
There is a list of languages at the bottom of the page on the welcome tab. Click the one you want.

C4Terminator, 2020-05-11:
Thanks 🙏

From the mobile office of
LEEKA Architecture and Planning
Neil Lee

closed 2020-05-11 by etracer65

Timlazisa, 2022-03-12:
Where at the bottom of the page??

Timlazisa, 2022-03-12:
[image]

McGiverGim, 2022-03-12:
Just under the Facebook button...

jonnytest1, 2024-12-11:
> There is a list of languages at the bottom of the page on the welcome tab. Click the one you want.

i would prefer moving that into a more visible spot maybe left of the version field in a dropdown 🤔

haslinghuis, 2024-12-12:
[image]