betaflight/betaflight-configurator #2023: Color selection and garish color resulting in blurry text

hydra, 2020-05-12:
See screenshots:

[image]

[image]

These used to be black, but some CSS styling has either changed accidentally or purposefully, but regardless of the reason for the change it results in blurry and messy text.

Version: 10.7.0-RC1
OS: Windows 10

mikeller, 2020-05-12:
Can you please provide screenshots of your change?

Asizon, 2020-05-13:
Really this blue text is more intuitive, I like alotof. For me is working so fine, 1080p and 125% escale:

[image]

McGiverGim, 2020-05-13:
I don't like the blue links neither. They seem like 90's. But I don't have time at this moment to propose a new solution. Maybe at some moment.

hydra, 2020-05-13:
@mikeller here's what 10.6.0 looks like. (built from git tag `10.6.0`)

[image]

[image]

You can remove the 'Missing Information' tag now.

mikeller, 2020-05-13:
@hydra:

Can you point out how, in your proposal, a user should be able to distinguish between emphasised text

[image]

and a clickable link

[image]

?

The usability of this is utterly useless. Proposals that ruin the usablity for the sole purpose of looks make no sense whatsoever and should not be considered.

hydra, 2020-05-13:
@mikeller The issue I have is that the colors/styling produces a garish result, and it appears there was an unintended change to the color of the text on the button to reset the Z axis, which is not a link, and shouldn't have a link style.

two fixes are needed:
* style the links so they are not garish.
* style the button so it's not using the link style and doesn't look like a link style.

With regards to how users can distinguish what is a link and what isn't, yes a visual clue is useful/needed; Suggest some other color that fits in with the Betaflight configurator color palette, not dark blue, which doesn't.  Perhaps an orange, or grey would work?

McGiverGim, 2020-05-13:
I didn't like the old black neither 🤪 Maybe I'm too exigent. 
I think some kind of grey can look good, but this needs play with it to search a good tone and see if it works.

mikeller, 2020-05-14:
@hydra: I frankly don't want to discuss matters of your taste here, and what you think is garish or not. Please stop wasting everybody's time.

closed 2020-05-14 by mikeller