betaflight/betaflight-configurator #2024: Rework annoying 'warning' popup in 10.7

hydra, 2020-05-12:
The new `Warning` pop-up is annoying.

[image]

Possible solutions:
1) Rename `Close` button to `Remind me on next connect` and add `Do not show again for this Flight controller` and perhaps `Never show this again` too.
2) Change it from an annoying pop-up to a UI element.  e.g. insert a 'Warnings' `div` panel between the buttons on the setup page and the 3D image below it.  only show the 'Warnings' div/panel when there are actually steps to be taken.
3) When 'Permanently enable Expert Mode' is enabled don't show the warnings.
4) Add an option to 'Ignore setup warning pop-up' and when enabled don't show the warning pop-up.

I'm open to other suggestions too.

Asizon, 2020-05-14:
@hydra If I have understood correctly, this warning window is to force users to repair the indicated issues, before starting a flight, as indicated in the text:
`Please fix these problems before attempting to fly your craft`

So really a `do not show again` check is not applicable to this, because is a mandatory security check before flight :)
I also think that it should also appear in expert mode. A msp compatibility message has also been added for future versions, so when we go to 4.3 with msp 1.44, we can see an update notice if we are using 10.7.0 and not 10.8. 0

mikeller, 2020-05-14:
@hydra: Happy to change this if you agree to personally deal with the resulting user requests based on users not getting this information and then complaining that something does not work, or else pay for somebody to answer these questions. Please state here if you agree to this and otherwise close this.

hydra, 2020-05-14:
@mikeller dont be silly. Of course I nor anyone else would agree to do that.

I'm raising this as a usability concern, not a personal attack, please treat this feedback appropriately.

hydra, 2020-05-14:
@Asizon yes, agreed some of the potential solutions I mentioned in the OP are less suitable.  The discussion here and in the BF slack has been useful.in understanding the reasons behind the popup and the messages being conveyed.

I do feel that some of the other solutions I mentioned would also work.

etracer65, 2020-05-14:
@hydra 

The entire point of the popup is **_to be annoying_**. So changing it to not be annoying would defeat the purpose. It has been shown time and again that the users do not read warning messages at the top of the page.

After the user is "annoyed" with this all they need to do is address the simple problems and then the warning goes away.

While this may be more "annoying" for us developers who are frequently flashing flight controllers (which I'm guessing is the basis for your complaint), that's secondary to benefits provided for the users and the reduction in ongoing support.

hydra, 2020-05-14:
yes, i understand the mainnreasons, i also offered  other ideas for those of us who are developers. that last thing we want to do is annoy contributors to the projects.  continually having to click close on the popup while working on firmware or the configurator itself is tiresome.

closed 2020-05-14 by mikeller