betaflight/betaflight-configurator #2028: 10.7-RC1 won't run on Ubuntu 18.04 LTS

frelar8, 2020-05-16:
Running 10.7-RC1 on Ubuntu 18.04 LTS appears not to work. Installs ok, but opening configurator only activates two processes but nothing happens. 10.6 and 10.5.1 works ok.

mikeller, 2020-05-17:
@frelar8: That's curious - configurator is working just fine on Ubuntu 19.10 here. Can you try completely removing and then re-installing configurator to see if this helps:

```
sudo dpkg -P betaflight-configurator
sudo rm -rf /opt/betaflight/betaflight-configurator

sudo dpkg -i ~/tmp/betaflight-configurator_10.7.0-RC1_amd64.deb
```

frelar8, 2020-05-17:
@mikeller: Tried removing and re-installing.
`sudo dpkg -P betaflight-configurator` also removes `/opt/betaflight/betaflight-configurator`
Also tried removing _betaflight-configurator_ from _.config_. Same as before; activates two processes and then nothing happens. Re-installing 10.6 (after removing 10.7-RC1) works fine

mikeller, 2020-05-17:
@frelar8: What window manager are you using?

frelar8, 2020-05-17:
@mikeller: I use GNOME Shell 3.28.4

haslinghuis, 2020-05-29:
Try
$ sudo mkdir /usr/share/desktop-directories/

Betaflight Configurator in UBUNTU 20.04 encounters a bug in XDG https://bugs.archlinux.org/task/33316

frelar8, 2020-05-29:
@haslinghuis Thanks for the tip but I can't try it, I've upgraded to 20.04 a few days ago (it works now).

closed 2020-07-05 by stale[bot]