betaflight/betaflight-configurator #2032: 10.7 RC1 won't detect FC if connected before starting the configurator

seriousrob, 2020-05-20:
BF Configurator 10.7 RC1 won't detect FC if configurator was opend after FC was connected to the PC.

Tested on DALRC F722 Dual and SpeedyBee F4 V2 with BF4.2 RC1, Windows 10 x64 Build 18362

Drivers are up to date and working properly.

Reproduce the issues:

Close the Betafkight configurator
Connect your FC to your PC and wait for it to boot up properly
Open the Configurator and try to select the COM port
Disconnect your FC
Reconnect your FC

Regards,
Robert

McGiverGim, 2020-05-20:
What we have observed is that for some reason it takes more time than usual to detect the ports. How much time you let for the Configurator to detect the port?

seriousrob, 2020-05-20:
Tried it again and left it open a bit longer. Can confirm what you've said. It detects the FC quite some time after start up. I was a little impatient probably. Thanks for the feedback.

closed 2020-05-20 by seriousrob