betaflight/betaflight-configurator #2041: osd issue in 4.1.7

capposteve, 2020-06-01:
Hi

ive noted an issue with the osd profiles, 
i set profile 1 fine, pressed save. switched to profile 2 and in the preview tab for profile 2. set up profile 2 differently. click save.
went back to profile 1 and it had moved my stuff to how i set up profile 2, i tried it the oposite way around and its done it again.

thanks

steve

McGiverGim, 2020-06-01:
You must explain it better, I'm unsure what you are doing exactly. Maybe screen captures?
It seems you are changing the PID profile, but this has nothing to do with the OSD profile. They are independent.
You have different checkboxes for each OSD profile, that you can assign as you want. And you can preview in the preview window selecting there what you want to see. But this has nothing to do with the upper PID profile.

haslinghuis, 2020-06-01:
That is expected. OSD Screen elements share position between profiles.
See also: [https://github.com/dronejunkie/betaflight/blob/master/docs/OSD%20Profiles.md](url)

capposteve, 2020-06-01:
> You must explain it better, I'm unsure what you are doing exactly. Maybe screen captures?
> It seems you are changing the PID profile, but this has nothing to do with the OSD profile. They are [...]

i have explained it correctly and no where in my message did i mentions pids. this is all to do with setting up different osd profiles, but really it does not let you, well not fully.

capposteve, 2020-06-01:
> That is expected. OSD Screen elements share position between profiles.
> See also: [https://github.com/betaflight/betaflight/blob/master/docs/OSD%20Profiles.md](url)

link does not work

McGiverGim, 2020-06-01:
I think I have understood now...

The POSITION of the OSD elements is global, it is not by profile. You can only select what elements do you want to be visible and what not in each profile, but the position is the same for all of them.

capposteve, 2020-06-01:
ok, why is this? as this is not how i want it set up, i thought each profile would be completely different?

haslinghuis, 2020-06-01:
Updated the link

haslinghuis, 2020-06-24:
Why is this issue still open? It does not make sense to have elements on other positions. That is not how our brain works.

closed 2020-08-01 by stale[bot]