betaflight/betaflight-configurator #2046: Mode switch command Angle - Acro -Horizont not working

cusglider, 2020-06-04:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
I use a switch to change between angle, acro and horizon mode via an aux channel. After configurator 10.7. RC2 and Beta 4.2 RC4 this no longer works, although with previous versions.

The configurator shows the input from the radio like usual, however there is no reaction from the FC. Not change of flight mode happens, no confirming beep after the switch.

2. Include ways to reproduce the problem;
Just set up an aux mode using the optical new 

3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
Matek F722SE
- the version of all the software used;
Configurator 10.7 RC2
Betaflight 4.2 RC4

- how things are connected / wired up.
soldered

4. It worked perfectly with Configurator 10.7 RC1 and Betfalfight 4.2 RC3 and previous versions
[image]
[image]

McGiverGim, 2020-06-04:
It is a bug at RC4, fixed in firmware in https://github.com/betaflight/betaflight/pull/9884

cusglider, 2020-06-04:
> It is a bug at RC4, fixed in firmware in betaflight/betaflight#9884 <https://github.com/betaflight/betaflight/pull/9884>

I tried the fix, however, at least with the F7 target it does not work and fix the problem

olliewells, 2020-06-05:
Having the same issue. BF 4.2

Describe your problem;
Previous versions of Bf (up to 4.1.7) allowed me to use a switch to swap between Angle, Horizon mode.

The modes are not available in the Modes section.

Include ways to reproduce the problem;
Set up radio receiver correctly, go to Modes, no available mode for Angle or Horizon.

Accelerometer is active. GPS is active. Airmode permantly on set to Off.

what hardware / software you are using;
Kakute F7 mini
Tekko 32 4in1 ESC

the version of all the software used;
Configurator 10.7 RC2
Betaflight 4.2 RC4

how things are connected / wired up.
Receiver is using Fport. 

It worked perfectly with Configurator 10.7 RC1 and Betfalfight 4.1.7

olliewells, 2020-06-05:
> I tried the fix, however, at least with the F7 target it does not work and fix the problem 

How do you apply the fix they mention?

olliewells, 2020-06-05:
> It is a bug at RC4, fixed in firmware in [betaflight/betaflight#9884](https://github.com/betaflight/betaflight/pull/9884)

Please can you give simple instructions on how to apply the fix

cusglider, 2020-06-05:
The video explains quite nicely

Test firmware:

betaflight_4.2.0_STM32F745_5d112384ae.zip
betaflight_4.2.0_STM32F7X2_5d112384ae.zip
betaflight_4.2.0_STM32F411_5d112384ae.zip
betaflight_4.2.0_STM32F405_5d112384ae.zip

(installation instructions: https://youtu.be/I1uN9CN30gw)

https://github.com/betaflight/betaflight/pull/9884

cusglider, 2020-06-05:
Follow this video instructions and check #9884 for links to the fix firmwarehttps://youtu.be/I1uN9CN30gw-- Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

cusglider, 2020-06-05:
I redid a clean install with the offered fix and indeed, it works!

My mistake was to copy the aux settings from a diff all file of RC3 and this apperently did not work with RC4 respectively the offered fix.

mikeller, 2020-06-06:
@cusglider:

> My mistake was to copy the aux settings from a diff all file of RC3 and this apparently did not work with RC4 respectively the offered fix.

This would not work, as the mode configuraton for ANGLE and HORIZON would be dropped in 4.2.0-RC3 - checking that all modes were properly configured in 4.2.0-RC4 would have revealed this. :wink:

Igorshp, 2020-06-11:
In the meantime, until next RC candidate is released, recalibrating accelerometer in configurator brings the modes back in the list and makes everything work as expected.

McGiverGim, 2020-06-24:
I suppose this is fixed... closing...

closed 2020-06-24 by McGiverGim