betaflight/betaflight-configurator #2049: PID sliders: Master=0.5, PDBalance=1.5 disables sliders

illmillrig, 2020-06-06:
if you set the Master multiplier to 0.5, and the PD Balance slider to 1.5 and save, the sliders disable themselves.

illmillrig, 2020-06-06:
it's not necessarily 0.5 and 1.5.  there are various combinations of slider values that when saved, cause the sliders to disable themselves.

illmillrig, 2020-06-06:
Master Slider 0.5
PD Balance 1.2
P and D Gain 1.3

illmillrig, 2020-06-06:
I think it would be better to manually have to disable the sliders via checkbox, in order to adjust individual RPY PID values, instead of guessing when i've changed those values and the sliders need to be disabled.

Asizon, 2020-06-07:
Seems that you are using outdated Configurator version like 10.6.0, for 4.2 please use latest 10.7.0 build

illmillrig, 2020-06-07:
[image]

for some reason I recall needing to update to 10.7 in order to flash 4.2, but could be mistaken

Asizon, 2020-06-07:
I got it! @illmillrig you need to provide more info about your setup, and diff. Anyway, I supose that your are making PID changes with Dmin feature disabled, im able to reproduce it in this way 👍 
[image]

Asizon, 2020-06-08:
From what I can see so far, its breaks in the new functionality for PD ratio calculation that is now aplied to D, but only on Dmin disabled part. Introduced in https://github.com/betaflight/betaflight-configurator/pull/1969

Asizon, 2020-06-08:
To reproduce:
Betaflight 4.2 RC4
Configurator 10.7.0 RC2
1-go to pidtuning tab
2-disable dmin feature
3-put master slider on 0.5
4-put PD Balance slider on 1.5
5-save

McGiverGim, 2020-06-09:
I have reproduced with the instructions of @Asizon, I reopen this again...

McGiverGim, 2020-06-09:
I let this open, is good to have this bug as a remember, maybe someone gets to a good fix.

haslinghuis, 2021-03-07:
Thanks for the testing procedure. It's still working now. Have to keep this in mind while we migrated to firmware sliders.

haslinghuis, 2021-06-03:
Guess this can be closed now as there are a lot of PR's merged to fix the legacy sliders and introduce firmware sliders. Now the firmware sliders needs improvement :smirk: 
https://github.com/betaflight/betaflight-configurator/pull/2429
https://github.com/betaflight/betaflight-configurator/pull/2418
https://github.com/betaflight/betaflight-configurator/pull/2415

closed 2021-06-03 by haslinghuis

ChristofferGylin, 2021-06-14:
I´m having the same issue, I don´t remember the values of the faders at the time, exept that the master was at 0.7, and the PD-ballance around 1.5. Dmin is turned off, and I´m on Betaflight 4.2 and running configurator 10.7.0

haslinghuis, 2021-06-14:
@ChristofferGylin this has been fixed (so far we know) in the upcoming releases.

PR #2053 (2020-06-09): Fixed PID tuning sliders disabling on certain values.
Fixes #2049. There is bugs, damn bugs, and rounding errors.