betaflight/betaflight-configurator #2051: Build configurator for arm64 device

ss8688, 2020-06-07:
I want to build it for Raspberry pi os (64bit),but I don t know how to build it for armv8 processor.

mikeller, 2020-06-09:
@ss8688: Betaflight configurator does not support the armv8 platform. There is an armv7 target, run `yarn gulp release --armv7`: https://github.com/betaflight/betaflight-configurator/blob/0ee367b84191a86c72cb96d3cab2d579b369f3f5/gulpfile.js#L104

closed 2020-07-18 by stale[bot]

smart2pet, 2023-02-01:
Please reopen this issue. I have the same problem.

smart2pet, 2023-02-10:
Ok.

godstargod, 2024-04-23:
> @smart2pet I wrote a _quick and dirty_ shell script available [here](https://raw.githubusercontent.com/romainvaltier/betaflight-configurator-arm64/master/install.sh) to automatically install and patch Betaflight-Configurator for my Pinebook-Pro (aarch64/arm64) and it is working properly.
> 
> `sudo ./install.sh`
> 
> You may have a try. No guarantee it will work for you as well.

thank you roman !! your work works off a pi400 with latest ubuntu 

now if i can only get to build edgetx native off of my pi ill be all set

cheers !

vu2iia, 2024-05-10:
Thank you very much [romain](https://github.com/romainvaltier). I installed it on my pi-4 2024-03-15-raspios-bookworm-arm64 & BFC v10.10, had been struggling to compile it. 
mahesh-vu2iia

moritzschaefer, 2024-08-23:
Nowadays Linux on Apple Silicon would very much appreciate if aarch64 would be supported

moritzschaefer, 2024-08-23:
Thanks for the info. Will have a look