betaflight/betaflight-configurator #2052: Add BF version to default filename when saving CLI content

fiechr, 2020-06-07:
Looking at a long list of saved _filenames_ from CLI output (like from a "diff all"), it's not easy to see which BetaFlight version generated the saved output, if the existing _default filename_ was accepted when saving the file.

I'd like to recommend adding the version number of the BetaFlight version the craft is using at the time, to the default filename the save to file dialog of the used OS will suggest.

For example, instead of just
`BTFL_cli_Tinyhawk...`
like this
`BTFL_417_cli_Tinyhawk_...`
or
`BTFL_v417_CLI_Tinyhawk_...`

This also would help mixing up different versions.

mikeller, 2020-06-08:
One problem I can see with this is that:
- when logging to SD card, we only have 8 characters for the filename - not enough to record much relevant information;
- when logging to onboard flash, the file name is only generated when the storage is mounted over MSC - at this point we have no way to tell what version of Betaflight has generated the log file - this could have happened before the last firmware update.

fiechr, 2020-06-08:
I'm not sure if we mean the same thing or if maybe I misunderstand what you mean?

I only meant the behavior of the "Save to File" button in the CLI tab of the Configurator. If I click on this button, I get a regular "File chooser" dialog to confirm or enter a (new) filename for saving the current content of the CLI window to. This dialog suggests a filename and this suggestion is what I think should be altered by adding the version info. As far as I understand this will be only saved using a regular Desktop OS like Windows, MacOS or Linux, which all have no issues with long(er) filenames.

mikeller, 2020-06-08:
@fiechr: Sorry, I misread what you said. Yes, doing this for CLI output in configurator can be done. In most cases you could just have a look at the file, or choose an appropriate file name yourself though.

fiechr, 2020-06-08:
True. Still, I'm a fan of good default values, since they aren't constraints but more of a suggestion. In my experience they get used more often than one anticipates, since most people are lazy when it comes to file management... :-)

closed 2020-07-18 by stale[bot]