betaflight/betaflight-configurator #2055: spamming save btn zeros all PID values

illmillrig, 2020-06-08:
clicking the save btn several times in a row sets all the PID values to 0.

haslinghuis, 2020-06-08:
Not able to reproduce with MatekF411/BF4.2RC4/BFC10.7RC2. Enabled/Disabled all sliders. Please provide information about configurator version, firmware version and maybe a `diff`, please let us know how to reproduce this issue.

illmillrig, 2020-06-08:
hmmm, i'm having problems reproducing now too.  But it will re-enable the sliders if you click the save button many times in a row very fast.

turn off DMin (not sure if this matters)
lower master slider to 0.7
lower pitch P by 1 (from 32 to 31) to turn off sliders
spam the save button by clicking it many times in a row, really fast

happens on all 3 of my quads.  BF 4.2 RC1. I can try RC2, but I imagine it's the same.

```
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Jun  3 2020 / 11:51:11 (9718914fd) MSP API: 1.43
# config: manufacturer_id: FPVM, board_name: BETAFLIGHTF4, version: f92ddc38, date: 2020-04-28T07:37:21Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name BETAFLIGHTF4
manufacturer_id FPVM
mcu_id 0023004d324d501020303256
signature 

# name: Beavis

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 35 2 1900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 450
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 30
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_p_ratio = 128
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set current_meter = ADC
set battery_meter = ADC
set yaw_motors_reversed = ON
set deadband = 3
set pid_process_denom = 2
set osd_warn_rssi = ON
set osd_vbat_pos = 14342
set osd_rssi_pos = 14356
set osd_tim_2_pos = 14349
set osd_warnings_pos = 14729
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set debug_mode = GYRO_SCALED
set vtx_band = 2
set vtx_channel = 5
set vtx_power = 3
set vtx_freq = 5809
set vcd_video_system = NTSC
set gyro_rpm_notch_harmonics = 1
set gyro_rpm_notch_q = 800
set gyro_rpm_notch_min = 90
set name = Beavis

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set anti_gravity_gain = 5000
set feedforward_transition = 100
set iterm_relax_type = GYRO
set p_pitch = 41
set i_pitch = 81
set d_pitch = 29
set f_pitch = 60
set p_roll = 38
set i_roll = 77
set d_roll = 27
set f_roll = 57
set p_yaw = 41
set i_yaw = 81
set f_yaw = 57
set d_min_roll = 0
set d_min_pitch = 0

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set anti_gravity_gain = 5000
set feedforward_transition = 100
set iterm_relax_type = GYRO
set p_pitch = 37
set i_pitch = 72
set d_pitch = 26
set f_pitch = 54
set p_roll = 34
set i_roll = 68
set d_roll = 24
set f_roll = 50
set p_yaw = 36
set i_yaw = 72
set f_yaw = 50
set d_min_roll = 0
set d_min_pitch = 0

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set anti_gravity_gain = 5000
set feedforward_transition = 100
set iterm_relax_type = GYRO
set p_pitch = 31
set i_pitch = 63
set d_pitch = 22
set f_pitch = 67
set p_roll = 29
set i_roll = 59
set d_roll = 21
set f_roll = 63
set p_yaw = 31
set i_yaw = 63
set f_yaw = 63
set d_min_roll = 0
set d_min_pitch = 0

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 28
set pitch_expo = 28
set yaw_expo = 28
set roll_srate = 56
set pitch_srate = 56
set yaw_srate = 56
set tpa_rate = 75
set tpa_breakpoint = 1200

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

haslinghuis, 2020-06-09:
Strange. After step 3 - changing P on Pitch from 32 to 31 (sliders disappear). I use my touchpad like a drumstick on the save button but still cannot reproduce? Please try RC4

illmillrig, 2020-06-09:
https://www.youtube.com/watch?v=xZV7C8y8PY4

Asizon, 2020-06-09:
I can reproduce it now, i can get this errors on debugging:
```
three.min.js:613 Uncaught TypeError: Failed to execute 'shaderSource' on 'WebGLRenderingContext': parameter 1 is not of type 'WebGLShader'.
    at Object.WebGLShader (three.min.js:613)
    at new <anonymous> (three.min.js:601)
    at THREE.WebGLPrograms.acquireProgram (three.min.js:611)
    at w (three.min.js:500)
    at THREE.WebGLRenderer.renderBufferDirect (three.min.js:547)
    at t (three.min.js:497)
    at THREE.WebGLRenderer.render (three.min.js:555)
    at Model.render (model.js:135)
    at Model.<anonymous> (model.js:85)
    at model.js:98
    at three.min.js:291
    at XMLHttpRequest.<anonymous> (three.min.js:286)
```

Asizon, 2020-06-09:
I have reproduce it on RC1 but no in RC4, @illmillrig please flash RC4 like @haslinghuis says and try again

Asizon, 2020-06-09:
Ok, I can reproduce also with RC4, same debugging errors like above mentioned

illmillrig, 2020-06-09:
just slap a timer on the btn to prevent spamming? :)

haslinghuis, 2020-06-09:
But who would spam a button anyway, You must be really inpatient :trollface:

illmillrig, 2020-06-10:
ahaha. yeah, I agree. I'm not sure how I stumbled across it

haslinghuis, 2021-06-05:
I'm sure you didn't find any :laughing:

closed 2021-06-05 by haslinghuis