betaflight/betaflight-configurator #2058: Can not reflash firmware offline (without internet connection).

and-sh, 2020-06-13:
When online firmware cache work well.

mikeller, 2020-06-14:
@and-sh: This is by design - the cache is designed to pull the latest information from the internet when the Firmware Flasher tab is selected. The only 'offline mode' that is currently supported is to manually select local files for the firmware and configuration that have been downloaded ahead of time, and flashing these.
Feel free to contribute the implementation of a better offline mode.

and-sh, 2020-06-14:
Thank you.
Is it possible to select the firmware from the cache as a local file?
Where is it stored on windows?

p.s.
Unfortunately I am not very strong in programming. 
It only comes to mind to make a local github mirror.

mikeller, 2020-06-14:
@and-sh:

> Is it possible to select the firmware from the cache as a local file?
Where is it stored on windows?

No, it is not - it is stored in the internal cache of NW.js.

But you can easily save the .hex file locally after selecting and loading the firmware in configurator, and showing the required configuration, and then hit 'save as file':

[image]

closed 2020-07-25 by stale[bot]