betaflight/betaflight-configurator #2060: Linux/amd64 deb package - Stable configurator's version 10.7.0 recognized as older than previous RC

dentex, 2020-06-14:
1. Stable configurator's version 10.7.0 recognized as older than previous RC
2. I already have installed configurator's v10.7.0 RC2:
```
2020-06-14 @ 10:27:32 -- Running - OS: Linux, Chrome: 80.0.3987.116, Configurator: 10.7.0-RC2
2020-06-14 @ 10:27:32 -- You are using an outdated version of the Betaflight Configurator.
```
Trying to install latest deb package, this error shows up:

[image]
translation:
```
Error: A newer version is already installed
...
Installed-Size in KiB not available
```
or, from a terminal:
```
$ sudo gdebi betaflight-configurator_10.7.0_amd64.deb 
[sudo] password:            
Reading package lists... Done
Building dependency tree        
Reading state information... Done
Questo pacchetto non è installabile [this package is not installable]
A later version is already installed
```
...while `dpgk` does it anyway [google translator]:
```
$ sudo dpkg -i betaflight-configurator_10.7.0_amd64.deb
dpkg: attention: demotion of betaflight-configurator from version 10.7.0-RC2 to 10.7.0
(Reading the database ... 389111 currently installed files and directories.)
Preparations to extract betaflight-configurator_10.7.0_amd64.deb ...
Extract of betaflight-configurator (10.7.0) on (10.7.0-RC2) ...
Betaflight-configurator configuration (10.7.0) ...
```

3. Info
-Computer-
```
Processor		: Intel(R) Core(TM) i7-4710HQ CPU @ 2.50GHz
Memory		: 16277MB (5253MB used)
Machine Type		: Notebook
```

-Operating System-
```
Kernel		: Linux 5.3.0-19-generic (x86_64)
Version		: #20~18.04.2-Ubuntu SMP Tue Oct 22 18:09:07 UTC 2019
C Library		: GNU C Library / (Ubuntu GLIBC 2.27-3ubuntu1) 2.27
Distribution	: Linux Mint 19.2 Tina
```

Please ignore this if the problem happens to be on my system and not in the deb package.
Thanks.

mikeller, 2020-06-14:
@dentex: Yeah, I noticed that one too - I don't think there is much that makes sense to do at this point - just use `dpkg` to install. This looks like a bug in the .deb library that is used for this - according to SemVer, `x.y.z` is newer than any `x.y.z-a` release candidate.

dentex, 2020-06-14:
Sure, no problem. Configurator installed and running.

I might check if `gdebi` has been already reported to show that error in case of similar version names.

closed 2020-06-14 by dentex