betaflight/betaflight-configurator #2061: Android App crashes on connect

Bradw-17, 2020-06-14:
-Samsung Galaxy S9 (Android 10)
-MatekF405CTR

When I connect the OTG cable I get a option to accept the connection, but after accepting the app quits.

ackzsel, 2020-06-15:
- LG V30 (h930)
- LineageOS 16.0 (Android 9)

Same issue. It asks for permission to connect and crashes after accepting. I attached a logcat from the event.
[logcat_06-15-2020_13-45-23.txt](https://github.com/betaflight/betaflight-configurator/files/4779980/logcat_06-15-2020_13-45-23.txt)

McGiverGim, 2020-06-15:
@MaximeAmadasi seems we start to receive feedback for your great work 😉

Bradw-17, 2020-06-15:
@McGiverGim I love the idea of having the full functionality of the configurator on my phone. Simplifies things so much. I just want to bring it to attention so it can be fully functional. Can't wait! :smile:

ackzsel, 2020-06-15:
Indeed. This would be AWESOME and I really hope it will work on our devices as well! Thanks all involved for bringing an Android app!

MaximeAmadasi, 2020-06-15:
@Bradw-17 @ackzsel Thank for your feedback. The permission can be ask two different ways : automatically after plug in the fc (with the app opened or not) or when clicking on the connection button of the configurator. So how many permission asking windows appeared before the app crashed ? And is the permission asking window responsible of the crash appeared before the app was launched or when the configurator was loaded ?

ackzsel, 2020-06-16:
I had the same results with OMNIBUSF4SD as @Bradw-17 described. The permission dialog was shown and after accepting the app crashed. 

BUT, I tried just a few minutes ago with a different FC (FURYF4OSD) and it worked! As soon as I get home I'll try again with the OMNIBUSF4SD to see if the problem is still present with that FC.

ackzsel, 2020-06-16:
@MaximeAmadasi, I might found something. HID_USB_CDC was set to ON on my OMNIBUSF4SD when it crashed the app. After setting HID_USB_CDC to OFF using my PC it was able to connect with the app! After setting it to ON again the app crashed again. I could reproduce this every time I tried. @Bradw-17 can you check if this might be your problem as well?

Bradw-17, 2020-06-16:
I'll take a look when I get home!

MaximeAmadasi, 2020-06-16:
@ackzsel This seems to be the issue. I was able to reproduce it. I think the issue is related to the Cordova plugin used for serial connection and not to the configurator code. I'll see what I can do to fix that !

WalcoFPV, 2020-06-18:
@ackzsel I changed my Github username, can you please change my username to WalcoFPV in your previous message ? Github didn't change it automatically. Thanks

ackzsel, 2020-06-19:
@WalcoFPV, done!

WalcoFPV, 2020-06-23:
@ackzsel @Bradw-17 I think, I solved the issue. Can you please try [this apk](https://drive.google.com/file/d/1WQl_tA9u75ByBWQtaIgos5EzbD-8ncHw/view?usp=sharing) You will have to uninstall the old apk before installing this one. Works well on my samsung s7-edge with aCL Racing F7 flight controller and `USB_HID_CDC  = ON`

Bradw-17, 2020-06-23:
I'll check as soon as I get home :+1:

Bradw-17, 2020-06-23:
Fix worked for me!

ackzsel, 2020-06-24:
@WalcoFPV, this fixed it for me as well! Thanks man.

WalcoFPV, 2020-06-25:
This issue will be fixed in future releases. Can be closed

McGiverGim, 2020-06-25:
We usually close the issues when the fix (in this case the original PR) is merged on master. I will link both.

scorpakascorp, 2020-06-28:
> @ackzsel @Bradw-17 I think, I solved the issue. Can you please try [this apk](https://drive.google.com/file/d/1WQl_tA9u75ByBWQtaIgos5EzbD-8ncHw/view?usp=sharing) You will have to uninstall the old [...]

Tnx!. It is OK now with this version. But before was same issue.
MATEKF405STD + Samsung Galaxy S8 Android 9.0

closed 2020-07-04 by mikeller

Delacius, 2020-07-22:
Can you post a link again, it says that apk is in the bin. I'm having same issue. Thanks.

Delacius, 2020-07-22:
> @ackzsel @Bradw-17 I think, I solved the issue. Can you please try [this apk](https://drive.google.com/file/d/1WQl_tA9u75ByBWQtaIgos5EzbD-8ncHw/view?usp=sharing) You will have to uninstall the old [...]

Can you give me this apk I'm having same issue. And can download it because it says file is in the bin.

Asizon, 2020-07-22:
[betaflight-configurator_10.8.0_android.zip](https://github.com/betaflight/betaflight-configurator/files/4962239/betaflight-configurator_10.8.0_android.zip)
@Delacius note that it is a development version and only for testing purposses

Delacius, 2020-07-22:
Thanks you so much, I'm on vacation no computer near by, so you are a life saver! 😉

All best

Goran