betaflight/betaflight-configurator #2062: Betaflight-configurator-installer_10.7.0_win32.exe not digitally signed

LTGFPV, 2020-06-14:
A issue that may keep new users from installing the new configurator is that during the download 

Windows Defender will stop the download because it detects the file as harmful, you have to select keep anyways. 

Norton Antivirus states that the file needs attention because the file is missing the following
Developer Name
Version
Fewer than 5 users that also use Norton have downloaded this file
File is very new
Unproven

Just wanted to give you a heads up.
Thanks for your continued support and hard work that you all put into this. 
Much appreciated,
Larry

# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

mikeller, 2020-06-14:
@LTGFPV:

> Fewer than 5 users that also use Norton have downloaded this file
File is very new
Unproven

I hope that these will resolve themselves once the release is more than a couple of hours old. :wink:

LTGFPV, 2020-06-14:
Yes, perhaps. Just wanted to make you aware of my findings.

Also tried out the android app, seems to work well on my Samsung Note 10G+, besides the alert stating that there is a newer release available, but I realize that the Android version is still being worked on. Still very cool. Thanks.

⁣Sent from BlueMail ​

closed 2020-07-25 by stale[bot]