betaflight/betaflight-configurator #2063: Not able to select any configuration pages

Johnnysod, 2020-06-14:
10.6.0 worked perfectly but 10.7.0 does not work at all.
It opens the initial screen when I connect to my Trashcan and I get a 3D box that animates with the drone orientation (In 10.6.0 it was a 3D model of a drone).
If I try to select any other pages down the left hand side then the program freezes and I have to close it.
I've tried reinstalling and re-downloading but no change.
I am using windows 10 on a fairly old HP EliteBook 8730
The drone is connected by usb (usb 2). It worked ok on previous versions and is able to read the gyro on the initial page.
I've reverted to 10.6.0

haslinghuis, 2020-06-14:
Could you please indicate which BF firmware version you are using?

McGiverGim, 2020-06-15:
True, usually this problem occurs when the firmware installed in the FC is not a final version, usually a beta version installed by the manufacturer.
Flash an official final version of Betaflight and the problem will be gone.

Asizon, 2020-06-15:
Hey, I think this is related to the graphic controller issue, solves disabling gpu usage

Asizon, 2020-06-15:
@Johnnysod try runing configurator with `--disable-gpu`

Johnnysod, 2020-06-15:
Hi
When I connect the drone using 10.6.0, the message under the BeataFlight
logo is

Configurator: 10.6.0
Firmware: BTFL 4.1.7
Target: HAMO/CRAZYBEEF4FS(STM32F411

I dare not update incase I can't configure it

John

______________________________________________________________________________

07764 335343 (Mobile)

0113 2841178 (Home)

TheJohnDyson@gmail.com (Home)

John.Dyson@bbc.co.uk (Work)

J Dyson, Riffa Manor Cottage, Off Harrogate Road, Pool in Wharfedale,
W.Yorks, LS21 2PT

rnxto, 2020-06-15:
the same problem. it happens to me in all my constructions. I use a 32bit pc but with 10.6.0 it works perfectly. I still didn't flash 4.2 on any.

Asizon, 2020-06-15:
Can you please try my avobe instructions?
Like that:
[image]

rnxto, 2020-06-15:
Ok.

Johnnysod, 2020-06-15:
Hi
Yes that worked. Thank you.

Do I have to add that to all the shortcuts (Start menu)?

The 3D model of a drone is back too, instead of a 3D box.

The dashes in your linked picture had me fooled so for anyone else with the same issue, just copy/paste this into your short cut properties Target. (Right click the Betaflight Configurator short cut/select Properties/Paste the text)

"C:\Program Files (x86)\Betaflight\Betaflight-Configurator\betaflight-configurator.exe" -disable-gpu

rnxto, 2020-06-15:
Perfecto!!

it worked. Thank you

TheIsotopes, 2020-06-16:
I knew that this was going to happen. 
@mikeller and I'm no longer the only one with this problem.

you should consider offering at least the nw.js version 0.45.6 as bugfix. this version has been running for weeks without problems.

Honi60, 2020-06-16:
Same issue **solve** by disable the GPU (prev version works without any problem)

CharlesSampson, 2020-06-19:
Also experiencing this, or a very similar issue. 

Steps to reproduce:
Connect FC to configurator, 
Setup tab looks totally normal (quad is an actual quad, not a box). 
Click to a different tab (Ports, config, etc)

Expected behavior: 
Tab with appropriate settings is opened

Actual behavior: 
I get a progress wheel with "Waiting for Data" that spins forever
(I am able to click back to Setup tab, so app isnt completely frozen)

----------------------

GPU disable has also not worked for me

Steps to reproduce: 
Connect FC to configurator (using --disable-gpu as instructed above)
Setup tab looks mostly normal but animation of quad is not there (instruments still responding as expected, just no animated quad in the center of screen). 
Click a different tab (Ports, config, etc)

Expected behavior: 
Tab with appropriate settings is opened

Actual behavior: 
App stays on Setup tab. The tabs all appear to be inert and do not respond to any clicks

mikeller, 2020-06-21:
@TheIsotopes: The reasons for not changing NW.js versions as a bugfix were explained to you in betaflight/betaflight-configurator#1989. Please stop badgering about it.

TheIsotopes, 2020-06-21:
@mikeller the bug with the buttons is only from 0.46.x
with 0.45.6 everything is ok in my opinion, i have been using this for weeks

alx-696, 2020-07-11:
I have the exact same issue with 10.7 and also with 10.8 nightly v20200711.695
With 10.6 everything worked fine; the issue occurred immediately after upgrading from 10.6. to 10.7 
"-disable-gpu" helped as a workaround.
Chrome is current.
Complete uninstall and reinstall of both BF Configurator and Chrome did not change anything.

d605, 2020-07-31:
> Same issue **solve** by disable the GPU (prev version works without any problem)

how do you disable the GPU? and where can it be found?

el-kwadro-fpv, 2020-08-24:
The solution with disabled gpu doesnt work for me either.

BF worked for month without any issue. I dont get it, why it wont anymore.

closed 2020-10-11 by stale[bot]

DJPOUT, 2021-02-06:
I did not try the method to exclude GPU but i tried first to right click on the icon of Betaflight Configurator ant click "troubleshoot program compatibility" or something like this, in french it's called "Résoudre les problêmes de compatibilité" and it works, then yo can save recommended settings. thank you to gave me the clue ;)