betaflight/betaflight-configurator #2065: Translation strings break UI

DusKing1, 2020-06-16:
Both zh_CN and zh_TW have this issue. Since we cannot shorten the translation, we hope we can find a solution in UI code.

I can't remember which version had the first problem, and I'm a little busy ATM. But I'm always willing to helping you guys find out.

[image]
[image]

McGiverGim, 2020-06-16:
This worked at some version? I don't remember any change to this, but if you think it worked before I can go back in the code trying to find the culprit. Version 10.6 was working?

DusKing1, 2020-06-16:
Also breaks on 10.6.0.

McGiverGim, 2020-06-16:
I think it never worked. Maybe in 10.5.0 it was not translated, I don't remember.
So we need to try to find a way to let bigger texts. I think I looked at it before and I didn't find a suitable way :(
I will try to look at it.

closed 2020-07-12 by mikeller