betaflight/betaflight-configurator #2067: 10.7 not opening correctly in Windows 10 64 bit

wdevine07, 2020-06-16:
Updated to latest version, and it just opens to a black screen. The mouse cursor changes as if their may be items there, but I can't click anything. I've attached a screenshot. Fully updated windows 10 64 bit. 10.6 still runs fine.

[image]

McGiverGim, 2020-06-17:
Could you try with the `--disable-gpu` parameter as explained here: https://github.com/betaflight/betaflight-configurator/issues/2063 ?

wdevine07, 2020-06-17:
> Could you try with the `--disable-gpu` parameter as explained here: #2063 ?

That worked! Thank you.

mikeller, 2020-06-21:
Duplicate of #2063.

closed 2020-06-21 by mikeller