betaflight/betaflight-configurator #2070: cannot open 10.7 in windows 8.1

masterZombie, 2020-06-19:
I have been using 10.6 with no problems. 10.7 will open a blank screen and a constant spinning next to the windows pointer like its trying but will never get past this. Tried to uninstal and reinstal but its the same problem. Running on windows pc 8.1. On second note it will install and run on my wifes laptop running windows 8.1. If 10.6 works but not 10.7, wouldnt that reflect a problem with 10.7?

McGiverGim, 2020-06-19:
Could you try with the `--disable-gpu` parameter as explained here: https://github.com/betaflight/betaflight-configurator/issues/2063 ?

masterZombie, 2020-06-19:
I have tried it and i get an error that says the target is invalid. its pretty unclear to me anyway exactly what to do. I am going to the betaflight shortcut and right clicking. Then selecting properties and inserting --disable-gpu at the end of the target path. Is this correct?

masterZombie, 2020-06-19:
Okay I read further in the thread and did a copy and paste. Now it works.

closed 2020-06-19 by masterZombie