betaflight/betaflight-configurator #2077: Some boxes around Configurator are broken(empty)

Asizon, 2020-06-22:
Betaflight 4.3 latest build
Configurator 10.8.0 latest build
Clean installation, no changes.
There are many checkbox boxes broken around the GUI, only appears checkbox button and the box is empty. Seems that it is happens after Jquery Update on PR https://github.com/betaflight/betaflight-configurator/pull/1995 . Here the first problem I can found:

[image]

McGiverGim, 2020-06-22:
It seems something is going wrong in the `i18n.localizePage();` function. It makes the HTML elements disappear. I will play a little more with it...

McGiverGim, 2020-06-22:
I have found the reason... but not the origin...

The problem is that we have this code:
[image]

But for some reason, when loading the page the tags have moved INSIDE the first span tag:
[image]

The localize REPLACES all the contents of the tag, and this is the reason why it disappears.
The funny thing is that if I change slightly the tag:
[image]

This does not happen anymore. It works as before. 

And this only happens with the new JQuery. Maybe a bug in this version? Some idea?

McGiverGim, 2020-06-22:
Tested JQuery 3.5.1 without luck...

Asizon, 2020-06-22:
@McGiverGim seems that the reasson is here, on changes part:
https://jquery.com/upgrade-guide/3.5/

McGiverGim, 2020-06-22:
True... let's see what is the better workaround for this...

McGiverGim, 2020-06-22:
@mikeller the changelog of JQuery shows the possibility to use "temporary" a plugin to restore old behaviour, but I suppose the way to go is to change self-closing tags that never "must" be self-closing, like `div, span, p` to the no self-closing form. If not this will bite use later.
What do you think? Is the way to go?

McGiverGim, 2020-06-22:
Pushed a PR with the changes. It seems to make all working again, but maybe I missed some part...

limonspb, 2020-06-28:
Maybe a related thing?
PWM settings disappeared starting from 
`Bump jquery from 3.4.1 to 3.5.0`
[image]

Asizon, 2020-06-28:
@limonspb yes, this bug is introduced by jquery update, it is already fixed but not yet merged

closed 2020-06-28 by mikeller