betaflight/betaflight-configurator #2082: Acronis Active Protection flagged 154 files in 10.7.0 with possible ransomware detected

AMainRCRacing, 2020-06-23:
I'm using WIN10 Pro. Acronis True Image 2019 (Build 17750) is my backup software. It has a feature Active Protection, which monitors for ransomware and illicit crytomining.

While trying to update BF 10.6 to 10.7.0 using the WIN32 .exe.download, Acronis Active Protection flagged 154 BF files for possible ransomware. The warning stated"Acronis Active Protection paused the program that modified your files. File content has been changed by a known suspicious pattern. - betaflight-configurator-installer_10.7.0_win32.exe - Unknown vendor". Then it asked me to Trust or Block. I chose Block and deleted the 10.7 files.

This is only the second time this warning has been activated. The first time, I resolved it. This time, it caught me totally by surprise. Total Protection removed the "offending" 154 files. I was able to reinstall 10.6 with no problems or warnings. I again tried to install 10.7.0 and the same warning came up except it was 165 files this time. I deleted all 10.7.0 files and reinstalled 10.6.0 with no problems or warnings. Unfortunately, the Acronis Total Protection report would not let me copy the "offending" filenames. 

On my third try, it flagged 159 files. I allowed Acronis to remove all BF files. I removed the block for the BF Configurator Installer in Acronis. In Windows File Manager, I navigated to the Program Files (x86) folder. I manually deleted the entire Betaflight folder along with all its subdirectories and files. This assured my computer was relatively free of Betaflight files. I did not modify anything to the Windows Registry. This is not my preferred method because of unexpected dependencies not being resolved by this "brute force" method of deleting a program and associated files, registry keys, etc.

I downloaded a fresh of copy BF 10.7.0. I scanned the installer with Windows Defender which showed no warnings. I started the BF 10.7.0 installer. One warning popped up the a .dat file could not be found so it could not proceed with uninstalling with what I presume to be a previous version. I allowed it to proceed with the installation. Everything went smoothly with no more warnings.

I'm not certain why I got the ransomware warnings when trying to update 10.6 to 10.7 on my first through attempts. When I did the clean install of 10.7 after deleting 10.6, no warnings appeared. Maybe Acronis Total Protection did not like the way 10.7 was modifying the 10.6 files. This above my pay grade! I just thought I'd alert the community to my problems upgrading from 10.6 to 10.7. It was only because of my familiarity with Windows that I was able to successfully upgrade to 10.7 on my particular PC.

haslinghuis, 2020-06-23:
This issue is related to not having a commercial digital certificate which will flag security products to issue false positives.

Duplicate of #2062

AMainRCRacing, 2020-06-24:
Thank you! If I find something, I'll try to search for other postings before I post. It would've saved me a heck of a lot of typing and your time to respond. Lesson learned!

closed 2020-06-24 by AMainRCRacing