betaflight/betaflight-configurator #2083: Betaflight 10.7.0 preview android.apk/prob.

jaisenas, 2020-06-24:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :). 
#
In the betaflight 10.7.0 preview android.apk
the vtx.js file should read as vtx: true not false.  It dose not display vtx setting at all.

McGiverGim, 2020-06-24:
> In the betaflight 10.7.0 preview android.apk
the vtx.js file should read as vtx: true not false. It dose not display vtx setting at all.

It works in the standalone computer Configurator? It only fails in the Android version?

The VTX says true when it is connected and configured. Have you configured it in the ports tab?

Asizon, 2020-06-24:
I think doesnt yet work file manager to load files like vtx table json

jaisenas, 2020-06-24:
ive setup and configure my vtx and flightcontroller proper yet it wont
display vtx as an option in the app. Insted i just set it in the CLI [set
vtx_power = 3] or [set vtx_channel = 0] or what ev

Asizon, 2020-06-24:
@jaisenas please provide a diff from your setup, seems that you are using a old version that not suported vtxtables

closed 2020-06-26 by no-response[bot]