betaflight/betaflight-configurator #2085: Connect button still says connect in OSX after configurator is connected.

SaturdayFPV, 2020-06-24:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
2. Include ways to reproduce the problem;
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
- the version of all the software used;
- how things are connected / wired up.
4. Remove this Text :).

1. Connect button still says connect in OSX after configurator is connected.
When trying to disconnect by clicking the connect button application will crash.
2. normally happens after you save your settings and restart the Flight Controller.
3. MacOS Mojave Version 10.14.6
    MAcBook Pro (13-inch, Mid 2012)
    Configurator 10.7 - Betaflight 4.2 
-- Running - OS: MacOS, Chrome: 80.0.3987.116, Configurator: 10.7.0
-- Loaded release information for configurator from GitHub.
-- Serial port successfully opened with ID: 1
-- MultiWii API version: 1.43.0
-- Flight controller info, identifier: BTFL, version: 4.2.0
-- Running firmware released on: Jun 14 2020 03:05:26
-- Board: DIAT/MAMBAF722(STM32F7X2), version: 0
    Connected via USB

Asizon, 2020-06-24:
Can you provide an screenshot of this issue?

SaturdayFPV, 2020-06-24:
may have to share vid via youtube since file size is an issue with this
account.

SaturdayFPV, 2020-07-25:
I have also noticed that the port defaults to CU instead of TTY in a lot of
cases.

closed 2020-08-31 by stale[bot]