betaflight/betaflight-configurator #2086: Vtx page broken if vtx table not available

rtlopez, 2020-06-24:
Hi fellows,

if device doesn't support vtx tables then vtx tab is not shown correctly, and configuration stop working.
On console you can get following error.

```
Uncaught ReferenceError: selectBand is not defined
    at HTMLSelectElement.populateChannelSelect (vtx.js:472)
    at HTMLSelectElement.dispatch (jquery.min.js:2)
    at HTMLSelectElement.v.handle (jquery.min.js:2)
    at Object.trigger (jquery.min.js:2)
    at HTMLSelectElement.<anonymous> (jquery.min.js:2)
    at Function.each (jquery.min.js:2)
    at S.fn.init.each (jquery.min.js:2)
    at S.fn.init.trigger (jquery.min.js:2)
    at S.fn.init.S.fn.<computed> [as change] (jquery.min.js:2)
    at initDisplay (vtx.js:216)
    at HTMLDivElement.process_html (vtx.js:39)
    at HTMLDivElement.<anonymous> (jquery.min.js:2)
    at Function.each (jquery.min.js:2)
    at S.fn.init.each (jquery.min.js:2)
    at Object.<anonymous> (jquery.min.js:2)
    at c (jquery.min.js:2)
```
probably following patch should be applied
```js
for (let i = 1; i <= TABS.vtx.MAX_BAND_CHANNELS_VALUES; i++) {
- selectBand.append(new Option(i18n.getMessage('vtxChannel_X', {channelName: i}), i));
+ selectChannel.append(new Option(i18n.getMessage('vtxChannel_X', {channelName: i}), i));
}
```
but I'm not quite sure. This fix works for me.
Old firmware might be affected or compiled without `USE_VTX_TABLE` flag.
I can make PR if you confirm solution.

Cheers ;)

McGiverGim, 2020-06-25:
I'm not too sure about the patch. If the problem happens when `USE_VTX_TABLE` is disabled, this part of code never must be executed, it is in the `else` part of `vtx_table_available`.

https://github.com/betaflight/betaflight-configurator/blob/be8d59e26f506b769f1702dd901a6055fe9bba71/src/js/tabs/vtx.js#L461-L474

So the problem seems related with `VTX_CONFIG.vtx_table_available` being `true` when it must be `false`. Or we are reading it bad in MSP from the FC, or the FC is sending it bad.

Can you give me more information about your setup to be able to reproduce it? Do you have a custom firmware without VTX_TABLE or is an old firmware before VTX_TABLE?

McGiverGim, 2020-06-25:
Sorry, too late in the morning... the code does what it must, goes to the else part because VTX_TABLE is not defined.

But I think the problem is that `selectBand` must be `selectedBand`, that is defined just before.

Can you try this change?

McGiverGim, 2020-06-25:
Again too early, forgot all my comments, bad solution, I need to reproduce it in my local, sorry for the spam...

McGiverGim, 2020-06-25:
After all of this I think your solution is right. It is populating the `channelSelected` so your patch seems ok.

Can you make the PR?

PR #2087 (2020-07-05): vtx tab fix for device without vtx tables
Fixes #2086 in vtx page if connected device doesn't support tables.
closed 2020-07-05 by mikeller