betaflight/betaflight-configurator #2089: Can spin motors from the Configurator src code regardless of MSP_SET_ARMING_DISABLED

limonspb, 2020-06-26:
Configurator must enable/prevent motors from spinning with
`mspHelper.setArmingEnabled(true, true);`
or
`mspHelper.setArmingEnabled(false, false);`

but it looks like this is not the case and we don't need to send this MSP command to be able to spin motors.
It means that Configurator can spin motors any time just by sending MSP_SET_MOTOR command, for example like that:
```
var buffer = [];

buffer.push16(1200);
buffer.push16(1200);
buffer.push16(1200);
buffer.push16(1200);
MSP.send_message(MSPCodes.MSP_SET_MOTOR, buffer);
```
just insert this code to any place in JS and it will spin motors regardless if arming is enabled or disabled.

McGiverGim, 2020-06-26:
I've verified the problem. I checked the Configurator part and all seems correct. It seems the firmware part is ignoring the disable arming flag. This is the firmware code:
https://github.com/betaflight/betaflight/blob/d85b372ae732acaaf514b1afac6940aefa5f2084/src/main/msp/msp.c#L2956-L2985

mikeller, 2020-06-26:
@limonspb:

> Configurator must enable/prevent motors from spinning with
mspHelper.setArmingEnabled(true, true);
or
mspHelper.setArmingEnabled(false, false);

Not sure what your reasoning is to state that 'configurator must prevent motors from spinning with a command that says it disables arming'. The command does exactly what it says - it disables arming.

This was introduced to prevent users from accidentally arming the craft *from the TX* when testing their TX setup with a battery plugged in (some craft need this to power the RX).
Disabling the effects of MSP commands that spin motors __with an MSP command__ seems pointless - if the client does not want to spin motors, they should not spin the motors in the first place.

McGiverGim, 2020-06-26:
True. Sorry @mikeller for all this noise, the user asked to me in the Slack chat first and I asked them to open an issue. If that is the intended way of work, I close this issue. 
If for some reason we want to change that, is a firmware request and not a Configurator one.

closed 2020-06-26 by McGiverGim

limonspb, 2020-06-26:
@mikeller thanks for explanations here and in slack.
Closing this issue.