betaflight/betaflight-configurator #2092: Profile doesn't save data properly.

effectic, 2020-06-27:
While setting up different values for each of the profiles, slider positions do not save.
For example for profile 1 I set filters sliders to the full right positions (2.0-2.0). For profile 2 I set them for (1.8-1.8).
After switching back to profile 1 I see different slider positions (2.0-1.8) or (2.0-1.5).
So changing profile settings affects other profiles.

Tried on bf 4.2.0 RC4 and 4.2.0 14-06-2020

McGiverGim, 2020-06-27:
The slider positions are not saved, only the pids. The Configurator tries to guess the slider positions based on that. The pids are the same that when saved? If true simply two combinations of sliders lead to the same values.

Asizon, 2020-06-29:
I have reproduced it now, its really weird. To reproduce:
-Go to filters tab
-Enable 2 sliders at 2.0x on the Profile 1
-Change to profile 2 and see weird values, they should be at default 1x but its wrong

McGiverGim, 2020-06-29:
@Asizon the PID values change between profiles or only the sliders?

Asizon, 2020-06-29:
Changes Gyro slider value with also values, but these showed values are not saved, checked in cli:
[image]

Asizon, 2020-06-29:
Maybe this is teh correct way to work? Seems that only Dterm filter depends of Profile?

McGiverGim, 2020-06-29:
Yes, the left tab is FOR ALL the profiles. Is common to them. "profile independent" says the title :P

Asizon, 2020-06-29:
Ouch!Please close this issue :)

effectic, 2020-07-01:
Still not clear. I suppose it should save all the parameters for each of the profiles.

McGiverGim, 2020-07-01:
As I said, the left filters are attached to all profiles. When you move the sliders, they are not stored, only the values for the filters, and when loading again it tries to find which sliders are ok for the values stored for the filters.
This is simply a bad decision design of the sliders. Nothing we can do at this moment. The sliders position must be stored for each profile in the FC and managed from there.

closed 2020-07-06 by mikeller