betaflight/betaflight-configurator #2095: Betaflight configurator offline mode

Ric-Alves, 2020-06-28:
This is my first pull request, so please be a bit tolerate for some wrong practice related with new pull requests.
I worked with "open source" engine management ECU (MSEFI) and it's GUI (TunerStudio) has a offline mode, which allows to "config" the ECU (by loading the ECU firmware of the MSEFI website) and when I connect to ECU, it compares with the config on ECU, it report the config aren't the same and asks if the user wants the laptop config or ECU config. If the user select laptop, it will upload the config to the ECU. If not, the GUI will replace its config with the ECU config.
This is a feature that I think it would useful to betaflight configurator.
The way I thought was, the possibility to load the defaults (selecting the FC hex file or the FC target) or to load the a saved diff file. This way, it is possible to work on the drone config without the need to be connected with the FC. For example to plan a new config/quad build or to do a small change quickly without the need to grab the quad and connect to the laptop.
Another situation I think it will be useful, when we are in the field and we use Speedybee or BF configuration App (which same times isn't easy to save a config or we are more interested to fly instead to save config) to change some parameters, when we connect the quad to the laptop, BF configurator will shows what were the changes we done on the field.
When the BF configurator has a config loaded, it is connected to the FC and the configs aren't the same (it could show a windows with the differences), it will prompt a message say something like " the FC config and GUI are different, do you want to use the GUI or FC config?"
If the user select GUI, it will upload the config to FC.
If the user select FC, the GUI replace its config with the FC config.

Sorry for my english, this is my 2 cent to help this great project.

JonnyPhenomenon, 2020-07-01:
this is a feature request, and one that I made a long time ago.  There are complications with implementing it, and its low on the priority list.  
https://github.com/betaflight/betaflight-configurator/issues/637

I just keep an old flight controller kicking around that I plug in when I want to explore the menus. it isn't a great solution, but it is better than nothing.

Ric-Alves, 2020-07-01:
@JonnyPhenomenon Your "post" is very similar to my idea. And I understand the devs problems and priorities.

I hope by doing another request for the same feature, and since your post was 2 years ago, maybe the necessary changes (parameters groups) were  done and it is possible to implement.

I only know about simple C language to microcontrollers. It isn't enough to contribute with code....

mikeller, 2020-07-06:
Duplicate of #637.

closed 2020-07-06 by mikeller