betaflight/betaflight-configurator #2096: VTX Page not working if VTX is set to soft serial

fiveseven808, 2020-06-29:
## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
- On only one of my quads using configurator 10.7.0, running BF 4.1.1 on a BetaFPV F411 3x uart + blackbox board, I am unable to view the VTX page when configuring the VTX port to a soft serial port. I get the following notice `Attention: Your VTX is not configured or not supported. So you can't modify the VTX values from here. This will only be possible if the flight controller is attached to the VTX using some protocol like Tramp or SmartAudio and is correctly configured in the Ports tab if needed.` 

Even after I've configured VTX tables via CLI, I still get this message when trying to use softserial vs a hard UART.

This is not the case on another one of my quads running BF 4.1.6  which is using VTX control over softserial as well.   
2. Include ways to reproduce the problem;
- Buy 3x UART F411 Betafpv whoop FC, 
- Setup softserial for the LED Pad
- Configure VTX to the softserial port
- Attempt to view the VTX tab
3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
 mentioned above
- the version of all the software used;
 Windows 10 v1903
 configurator 10.7.0
 BF 4.1.1 
- how things are connected / wired up.
 USB to FC

Asizon, 2020-06-29:
Working fine for me with your steps, please provide a diff and resource from your setup

McGiverGim, 2020-06-29:
This message appears when the FC returns 0 (USUPPORTED) or 255 (UNKNOWN) when asking for the VTX information. What kind of VTX have you configured? If it is a whoop, the most probably is that the VTX is a camera/vtx all in one without connection to the FC to control it.

fiveseven808, 2020-06-29:
Ahhh shoot, my apologies. I had configured the softserial UART to the wrong pin... I configured it to the LED vs LED_STRIP and betaflight wasn't having any of it. 

I reconfigured to pin A08 and I can now access to VTX Page now. Thanks for the explanation McGiverGim! I didn't realize that those smart audio protocols were bi-directional. It makes sense now how a sanity check is performed. 

Thanks guys!

closed 2020-06-29 by fiveseven808